ZoneMinder/zoneminder #2078: ZoneMinder 1.32.0 release punchlist

knnniggett, 2018-04-17:
The following is a list of punchlist items we need to complete before we release zoneminder 1.32.0:

- @knnniggett fix merge conflicts in #2065 
- @knnniggett resolve #2067 - I was waiting for the Crud project to merge https://github.com/FriendsOfCake/crud/pull/582 I may have to do something hackish in the meantime
- everyone - verify a minimal amount of documentation exists for @connortechnology 's new features
- everyone - beta test, beta test beta test. There are a lot of new features so we need to test the heck out of this
- @knnniggett @connortechnology review open pull requests to determine eligibility for next release

Please provide additional bullet points

connortechnology, 2018-04-17:
The Storage listing in the navbar needs work. It currently shows the value from a db table which is the sum of the disk space taken by each event. It is NOT the actual disk space used/available.  It needs to show both pieces of information and be colour coded. 

There is an issue that I am investigating where the monitor thinks it has lost signal and it goes back and forth between lost/reacquired. There is no accompanying logs saying why.  

There an issue where sometimes if a monitor fails to connect to the camera the db log entry fails.  

I intend to add another field to filters to specify whether to run the filter on that Server.  

I intend to add another option for filtering to specify whether the storage area is the one that the event is stored on, versus the one that the Monitor is assigned to.

Cleanup css skin themes.  Restore classic to something more like the original look.

merge @pliablepixels ptz update

knnniggett, 2018-04-20:
In addition to the typical stuff I put in the release notes, I'm going to show a summarized list of new features. 

Here is what I have so far:
- Record to mp4 container files
- Record to multiple storage folders
- Web UI rewrite
- Montage Review rewrite (?)
- All system specific path variables are now stored in config files under a conf.d folder
- Add an arbitrary website as a non-recordable monitor

What else did I miss?

pliablepixels, 2018-04-20:
On the webUI rewrite if you are referring to the storage areas UI work, I think "Web UI cleanup" would be more appropriate

knnniggett, 2018-04-20:
The web ui rewrite I am referring to is @connortechnology bringing in Kyle's changes shown in #1665 
e.g. all the menu items are down the side now, and popups are minimized.
At least that is what I think happened.

connortechnology, 2018-04-20:
Scalability improvements, can now handle > 100 cameras
Ability to move events from one place to another
raspberry pi OMX hardware h264 decoder support
disk space usage in console
event thumbnails now play video on mouseover
dragndrop sorting of monitors

knnniggett, 2018-05-04:
\### Documentation<br>
\- [ ] Wiki – Update Wiki documentation to reflect any changes to existing features<br>
\- [ ] Local – Update internal documentation to reflect any changes to existing features<br>
\- [ ] Wiki – Add new documentation to Wiki for any new features<br>
\- [ ] Local – Add new internal documentation for any new features<br>

\### Build<br>
Tested with the following build method:<br>
\- [ ] AutoTools<br>
\- [ ] CMake<br>

\### Distro<br>
These tests were performed on the following Linux distributions:<br>
\- [ ] Ubuntu LTS<br>
\- [ ] Ubuntu<br>
\- [ ] CentOS/Redhat<br>
\- [ ] Fedora<br>
\- [ ] Other<br>

\### Browser<br>
These tests where performed using recent versions of the following browsers:<br>
\- [ ] FireFox<br>
\- [ ] Chromium<br>
\- [ ] Chrome<br>
\- [ ] Internet Explorer<br>
\- [ ] Safari<br>
\- [ ] Mobile (state browser)<br>
\- [ ] Other<br>

\### Startup<br>
\- [ ] Verify initscript can auto-start ZoneMinder during the boot process<br>
\- [ ] Verify initscript can manually stop, start, and restart ZoneMinder<br>

\### Filters<br>
\- [ ] Verify a filter can be manually executed<br>
\- [ ] Verify a background filter runs automatically<br>

\### Authentication<br>
\- [ ] None - Trivial<br>
\- [ ] Plain – Set authentication to plain and verify<br>
\- [ ] Hashed – Set authentication to hashed and verify<br>
\- [ ] Remote – Set authentication to remote and verify<br>

\### Streaming Methods<br>
\- [ ] Local – Verify successful stream from this camera type<br>
\- [ ] Http – Verify successful stream from this camera type<br>
\- [ ] Ffmpeg – Verify successful stream from this camera type<br>
\- [ ] Libcurl – Verify successful stream from this camera type<br>
\- [ ] Libvlc – Verify successful stream from this camera type<br>

\### Recording Functions<br>
\- [ ] Monitor - Verify this function works<br>
\- [ ] Modect - Verify this function works<br>
\- [ ] Record - Verify this function works<br>
\- [ ] Mocord - Verify this function works<br>
\- [ ] Nodect - Verify this function works<br>

\### PTZ Control<br>
\- [ ] Verify ptz control works<br>

\### Skins<br>
\- [ ] Classic – verify field alignments<br>
\- [ ] Flat – verify field alignments<br>

\### Log Files<br>
\- [ ] Inspect log files for anything unusual<br>

\### Upgrade<br>
\- [ ] zmupdate - Verify an upgrade for a recent minor release<br>
\- [ ] zmupdate – Verify an upgrade for a past major release<br>

\### Optional Features<br>
\- [ ] X10 - verify basic communication with an x10 device<br>
\- [ ] E-mail – verify events are sent through e-mail<br>
\- [ ] Upload – verify events are uploaded<br>
\- [ ] Cambozola – verify from Internet Explorer<br>

\### API<br>
\- [ ] Verify that you can query monitors and events

MalakymR, 2018-06-07:
I had sorted some of these as they came up but lost track.

Some of the other new dependencies were mostly perl modules - pecl-apcu, file-slurp, format-human-bytes

knight-of-ni, 2018-09-12:
Released!

closed 2018-09-12 by knight-of-ni