ZoneMinder/zoneminder #2079: GIT from 17.4.2018 at 14:48 GMT+2

zd59, 2018-04-17:
Tried to compile Zoneminder from GIT:
git clone https://github.com/ZoneMinder/ZoneMinder.git
at 17.4.2018 at 14:48 GMT+2

At compilation the system complains about zero file lengths and empty subdirectories.
Log lines are at the bottom.
I checked and all is true - there are a missing subdirectories and zero length files in git.

Pleas fix this.

**Relevant log lines:**
```
WARNING:  zero length file web/skins/classic/css/classic/views/zone.css
WARNING:  zero length file web/skins/classic/css/dark/views/zone.css
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Vendor/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Lib/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Controller/Component/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Console/Command/Task/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Console/Templates/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/webroot/files/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/webroot/js/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Test/Case/Controller/Component/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Test/Case/Model/Behavior/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Test/Case/View/Helper/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Test/Fixture/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Plugin/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Locale/eng/LC_MESSAGES/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Model/Behavior/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/Model/Datasource/empty
WARNING:  zero length file web/api/lib/Cake/Console/Templates/skel/View/Scaffolds/empty
WARNING:  zero length file web/api/lib/Cake/Test/test_app/Controller/Component/empty
WARNING:  zero length file web/api/lib/Cake/Test/test_app/Console/Command/Task/empty
WARNING:  zero length file web/api/lib/Cake/Test/test_app/Plugin/TestPlugin/Console/Templates/empty
WARNING:  zero length file web/api/lib/Cake/Test/test_app/Plugin/TestPluginTwo/Console/Command/Task/empty
WARNING:  zero length file web/api/lib/Cake/Test/test_app/Plugin/TestPluginTwo/Console/Templates/empty
WARNING:  zero length file web/api/lib/Cake/Test/test_app/tmp/empty
WARNING:  zero length file web/api/lib/Cake/Test/test_app/View/Scaffolds/empty

```

knnniggett, 2018-04-17:
You didn't tell us *how* you are trying to build from git.
This matters because, for example, this issue in indeed already fixed for the rpm packages.

zd59, 2018-04-17:
Hi!

Distribution: Slackware64-14.2
Compilation direct from git sources, I added compile script. Remove ".txt" at the end.
The previous versions of GIT source (a month ago) compiled & installed & run excellent.

[ZoneMinder.SlackBuild.txt](https://github.com/ZoneMinder/zoneminder/files/1920059/ZoneMinder.SlackBuild.txt)

knnniggett, 2018-04-17:
Okay, that was helpful. I recall working with Bob when he made that script.
At line ~79 there is a find command that is setting permissions. Add a second find command that does the same thing we are doing in the rpm packaging scripts: https://github.com/ZoneMinder/zoneminder/blob/master/distros/redhat/zoneminder.spec#L171

Replace the buildroot rpm macro with "."

Note that this will also delete the .htaccess files. 
For rpm packaging I migrated the contents of those .htaccess files to the main config file. 
See -> https://github.com/ZoneMinder/zoneminder/blob/master/distros/redhat/apache/zoneminder.conf.in#L47

You will either need to make the same change yourself, or modify the find command referenced earlier to *not* delete .htaccess files. Your choice.

I do plan to make this global across all builds at some point, but it is not high on my priority list.

closed 2018-04-21 by knnniggett