ZoneMinder/zoneminder #2084: mocord and bulk frames just don't work, small events are recorded, many frames are lost

marcmerlin, 2018-04-20:
**ZoneMinder Version (`zmaudit.pl -v`):**
gargamel:~# zmaudit.pl -v
1.30.4
Debian Source: zoneminder-dmo
Version: 2:1.30.4-dmo4

Many details were posted on 
https://forums.zoneminder.com/viewtopic.php?p=105498

I'm supposed to get events of 10mn, unless there was an alarm. In real life, I'm getting totally weird event lengths and interval between frames that can be 10, sometimes 60 seconds+ !
Basically it seems that mocord was not tested and does not work :-/

Excerpt from the post referenced above:

All frames are marked as normal, so they were not bulk'ed. Yet, I get 10 seconds between frames, and then it jumps from frame 25 to 55 with a 45 second jump. What's going on? 

Frame Id	Type	Time Stamp	Time Delta	Score
Frames - 5076676
(frame1-19 cut out by me)
20	Normal	23:09:35	10.24	0
21	Normal	23:09:35	10.88	0
22	Normal	23:09:36	11.43	0
23	Normal	23:09:37	11.97	0
24	Normal	23:09:37	12.38	0
25	Normal	23:09:37	12.88	0
55	Normal	23:10:10	45.85	0

then the next event is exactly 2 frames. 2 frames only, really?
Frames - 5076692
Frame Id	Type	Time Stamp	Time Delta	Score
1	Normal	23:10:01	0.00	0
2	Normal	23:10:14	12.69	0

But wait, 23:10:01 < 23:10:10, so the next 2 frame only event overlaps with the previous one.

Then the next event looks like this:
Frames - 5076695
Frame Id	Type	Time Stamp	Time Delta	Score
1	Alarm	23:10:24	0.00	92
2	Alarm	23:10:24	0.10	92
3	Alarm	23:10:25	0.21	90
4	Alarm	23:10:25	0.32	88
5	Alarm	23:10:25	0.43	90
6	Alarm	23:10:25	0.65	89

6 frames, all alarm, and then it jumps to the next event that 159 frames, and after a 2mn gap, which is a big deal again.
Frames - 5076712
Frame Id	Type	Time Stamp	Time Delta	Score
1	Normal	23:12:14	0.00	0
2	Normal	23:12:15	0.48	0
3	Normal	23:12:16	1.10	0
4	Normal	23:12:16	1.51	0

mccrobie2000, 2018-05-05:
Sorry, left a comment in the other Issue #1698

marcmerlin, 2018-07-05:
I worked around this problem by having events last 10mn and not create new events even if mocord found an alarm.
Still broken, but I found a way around it

asantaga, 2018-07-20:
@marcmerlin Can u explain your workaround?

marcmerlin, 2018-07-23:
@asantaga , what I wrote "I worked around this problem by having events last 10mn and not create new events even if mocord found an alarm."

xztraz, 2018-09-20:
Also having problems with mocord.  low rez video records as it should where zone detection is active. this should trigger the mocord stream (high rez) byt i get alot less events on the hig rez stream. sometimes events are completly missing. restart usualy makes stuff a bit better. 

buffers and such set the same for low res and high res.

[image]

i wish there where a prettier integration of low and high resolution streams since almost all ipcameras support this and it's kind of the only way to use many cameras without overloading the cpu.

marcmerlin, 2018-11-19:
still a problem

marcmerlin, 2019-01-19:
still a problem

marcmerlin, 2019-03-20:
still not stale

marcmerlin, 2019-05-21:
still not stale
Sigh

connortechnology, 2019-05-21:
Have you tested with latest master?  Not that much has changed around this code, but there HAVE been changes.  1.30.4 is pretty old.

connortechnology, 2021-02-05:
Please report if this is still an issue with 1.34.23