ZoneMinder/zoneminder #2089: getAuthHash API issues

pliablepixels, 2018-04-29:
@connortechnology, need to revisit how getAuthHash API works

1. When OPT_AUTH is enabled, it returns an auth_hash token
2. When OPT_AUTH is disabled, the API throws an error `"name": "Call to undefined function generateAuthHash()",` --> should it not return auth_hash = ""  or something like that?
3. How will this API react if a user has "&user=&pass=" ? I forgot what setting of auth results in this output so I can't test it.

I plan to use this new API if ZM version > 1.31.41, so I need to ensure its as predictable as scraping the auth key from the watch page (which is what I am doing now)

pliablepixels, 2018-04-29:
Ok, I think this API should basically replicate what functions.php is doing:
```
if ( ZM_OPT_USE_AUTH ) {
  66     if ( ZM_AUTH_RELAY == 'hashed' ) {
  67       $args[] = 'auth='.generateAuthHash( ZM_AUTH_HASH_IPS );
  68     } elseif ( ZM_AUTH_RELAY == 'plain' ) {
  69       $args[] = 'user='.$_SESSION['username'];
  70       $args[] = 'pass='.$_SESSION['password'];
  71     } elseif ( ZM_AUTH_RELAY == 'none' ) {
  72       $args[] = 'user='.$_SESSION['username'];
  73     }
  74   }
```

fennec622, 2018-04-29:
Hi

if i understand now i can't use user pass in api request from watch page or snapshot ?

thanks

pliablepixels, 2018-04-29:
@fennec622  this is a new API which never existed before. Its not related to the watch page.

connortechnology, 2018-04-30:
I agree it should return '' if auth is disabled.  I will be merging a fix later today. 

I do not agree that it should return user=&pass= at least not how it is named, etc.

closed 2018-05-02 by pliablepixels