ZoneMinder/zoneminder #2092: zms logs invalid-HTTP output "Not logging to file because level <= mFileLevel"

johnruemker, 2018-05-01:
**Issue Description**

When zms is queried for a monitor's jpeg, it first logs "Not logging to file because level <= mFileLevel" if LOG_LEVEL_FILE is set to anything less than INFO.  This breaks http parsing, such as by aiohttp.  

With LOG_LEVEL_FILE at WARNING: 

~~~
# curl --insecure -XPUT 'https://zm.example.com/cgi-bin-zm/nph-zms?mode=single&buffer=1000&monitor=4&user=example&pass=test' --output curl.out
# head -n 12 curl.out
Not logging to file because level <= mFileLevel
HTTP/1.0 200 OK
Server: ZoneMinder Video Server/1.31.43
Expires: Mon, 26 Jul 1997 05:00:00 GMT
Last-Modified: Tue, 01 May 2018 12:34:17 GMT
Cache-Control: no-store, no-cache, must-revalidate
Cache-Control: post-check=0, pre-check=0
Pragma: no-cache
Content-Length: 443425
Content-Type: image/jpeg

JFIFC
~~~

With LOG_LEVEL_FILE at INFO:

~~~
# curl --insecure -XPUT 'https://zm.example.com/cgi-bin-zm/nph-zms?mode=single&buffer=1000&monitor=4&user=example&pass=test' --output curl.out
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  420k  100  420k    0     0   420k      0  0:00:01 --:--:--  0:00:01 2923k
[root@mothership ~]# head -n 2 curl.out 
JFIFC

~~~

This was a problem for me because it breaks home-assistant's zoneminder platform began being unable to serve up my camera feeds.   

This began sometime after upgrading to zoneminder 1.31, which I did not long ago.  I'm not sure if it occurred with all 1.31's or only started recently - I wasn't paying much attention to home-assistant while I worked on switching to 1.31.  

**Workaround**

Under Options->Logging in the web UI, set LOG_LEVEL_FILE to INFO. 

**ZoneMinder Version (`zmaudit.pl -v`):**

~~~
# zmaudit.pl -v
1.31.43          
# rpm -q zoneminder
zoneminder-1.31.43-1.0.20180427gitg86b2f6a12.fc27.x86_64
~~~

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

No

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**

Fedora 27, via a self-rolled container.  

~~~
# cat /etc/fedora-release                
Fedora release 27 (Twenty Seven) 
~~~

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**

sv3c bullet cams. 

**Relevant log lines:**

In the ssl_access_log when home-assistant makes these queries: 

~~~
192.100.0.4 - - [01/May/2018:08:20:45 -0400] "GET /cgi-bin-zm/nph-zms?mode=single&buffer=1000&monitor=6&user=example&pass=test HTTP/1.1" 200 -
~~~

connortechnology, 2018-05-01:
Thanks for the excellent report.

I have pushed a fix removing the code that outputs that line. It was only there for debug purposes.

knnniggett, 2018-05-04:
Ohhhh.... another Fedora user.  

Did any of the following new features cause issues with your upgrade?
- zm/conf.d folder
- new content in zm apache config file
- (new) zm apache config file no longer automatically placed into httpd/conf.d (see README for details)

johnruemker, 2018-05-04:
~~~
Did any of the following new features cause issues with your upgrade?

zm/conf.d folder
~~~
Yes, I ran into this as a problem.  I was unaware of the conf.d change when I upgraded. IIRC starting via zmpkg.pl was throwing an error of some kind and I was led to the config file, which showed me I needed to convert my 1.30 config to conf.d style.  

It would be nice if maybe the Fedora RPM post-script could alert the user they need to convert if it finds a modified zm.conf, or something like that.

~~~
new content in zm apache config file
~~~
I dont think I had any problems with this.  

~~~
(new) zm apache config file no longer automatically placed into httpd/conf.d (see README for details)
~~~
This also bit me.  I actually didn't see any note in the README about this one - I'll have to look for it now.  It took me awhile to realize why Apache wasn't serving anything but the default fedora index and couldnt find ZM's root contents.  

Seems like the Fedora RPM could link or copy the config into place to address this.  

I'd be happy to help with anything you're interested to have checked or considered more closely on Fedora.

Thanks!

knnniggett, 2018-05-04:
The release notes will notify users of the new use of the conf.d folder, so you're right. Today this is a non-documented feature in our development branch. 

As you mentioned, RPM doesn't touch zm.conf if you have modified it. It currently does tell you this in its output with the message "saving zm.conf as zm.conf.rpmnew" or something like that. 

The problem is that, you have to watch the rpm output like hawk or you won't see it.  Unlike debian, rpms cannot be interactive, so there is no good way to pause the package installation with a prompt for the end user to look at something.  

The good news is that, once you move any custom config changes into your own custom config under the conf.d folder, you should not see this problem with zm.conf ever again. This very issue was what drove me to add support for the conf.d folder in the first place.

Here is the README source:
https://github.com/ZoneMinder/zoneminder/tree/master/distros/redhat/readme

Installation Step 6 discusses how to handle the zm apache config
Upgrade Step 3 discusses how to handle the zm apache config for existing users upgrading their system.

In case you are wondering what caused this change see this bugzilla:
https://bugzilla.rpmfusion.org/show_bug.cgi?id=4861

The end user was a new ZoneMinder user with an existing web server running other stuff. He didn't realize the rpm auto-installed the apache config file, which caused the web server not to start until he followed the README to set up ZoneMinder. I agree with his argument that this should not be an automatic thing.... so consequently the end user must now manually symlink the zm apache config into place. That way the web server config does not get changed until the end user knows about it and is ready.

In our drive to release the next version of ZoneMinder, what we need are people from all distros to go through all the basic stuff to make sure are no new issues.... just normal day-to-day activities like adding a monitor, deleting a monitor, make sure motion detection works, filters work, etc. Here is a punchlist of general things we look at: https://github.com/ZoneMinder/zoneminder/issues/2078#issuecomment-386674189

If you are willing, it would be big help to test this.

closed 2018-05-18 by connortechnology