ZoneMinder/zoneminder #2104: Events view no longer shows highest-scoring frame

kimmeld, 2018-05-12:
I just upgraded ZoneMinder (from zmrepo-testing) and have found that the event list view now shows the first frame of the event, not the highest scoring frame.  This makes reviewing the events more time consuming.

All of my cameras are using Video Storage with the following options:

- Save JPEGs = Analysis Images only (if available)
- Video Writer = X264 Encode
- Optional Encoder Parameters = None
- Store audio stream = No

**ZoneMinder Version (`zmaudit.pl -v`):**

```
# zmaudit.pl -v
1.31.43
```

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

```
# rpm -qi zoneminder
Name        : zoneminder
Version     : 1.31.43
Release     : 1.0.20180426gitg86b2f6a12.el7.centos
[snip]
```

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**

```
# cat /etc/redhat-release
CentOS Linux release 7.5.1804 (Core)
```

kimmeld, 2018-05-19:
I think I have some further information about what's going on.  Here's my analysis of the problem:

1)  The events view uses the snapshot image as the source.

See https://github.com/ZoneMinder/zoneminder/blob/77b144a13e191614f6ac5036338119947e29213b/web/skins/classic/views/events.php#L221
and https://github.com/ZoneMinder/zoneminder/blob/77b144a13e191614f6ac5036338119947e29213b/web/includes/Event.php#L318-L319

Basically, the events list calls $event->getThumbnailSrc(), which generates a URL that eventually retrieves snapshot.jpg from the event directory.  The comment at the top of this function indicates that the snapshot should be the image with the most motion.

2)  zm_event.cpp only writes out the first event frame to snapshot.jpg

There are two places where snapshot.jpg is written:

AddFrameInternal writes it for the first frame:  https://github.com/ZoneMinder/zoneminder/blob/77b144a13e191614f6ac5036338119947e29213b/src/zm_event.cpp#L462-L470  This seems to be just used in the code that adds pre-alarm frames.

AddFrame writes it for the first frame as well:  https://github.com/ZoneMinder/zoneminder/blob/77b144a13e191614f6ac5036338119947e29213b/src/zm_event.cpp#L522-L526

I think we also need some code around here https://github.com/ZoneMinder/zoneminder/blob/77b144a13e191614f6ac5036338119947e29213b/src/zm_event.cpp#L592-L593 to write out the sanpshot image again.  I think the changes would look something like this:

```
    if ( score > (int)max_score ) {
      max_score = score;
      char snapshot_file[PATH_MAX];
      snprintf(snapshot_file, sizeof(snapshot_file), "%s/snapshot.jpg", path);
      WriteFrameImage(image, timestamp, snapshot_file);
    }
```

This will overwrite snapshot.jpg with the frame whenever a new highest-scoring frame is seen, which seems to be what the rest of the code is expecting to happen.

Unfortunately, I don't have an environment set up to build ZoneMonder from source, so this is where I'm stuck.  Hopefully someone can take this and run with it.

connortechnology, 2018-05-22:
You are absolutely correct, and I could have sworn that I made that exact code change a while ago.  However I don't see it, so I will re-commit it.

closed 2018-06-08 by connortechnology