ZoneMinder/zoneminder #2105: PHP Errors when setting up filters

kimmeld, 2018-05-12:
I just upgraded ZoneMinder (from zmrepo-testing) and have found that the filters page no longer works.  Filters themselves appear to do their thing in the background, I just can't maintain them or use them through the front end.

There are two PHP errors that are happening, both appear in ssl_error_log:

1. PHP Fatal error:  Can't use method return value in write context in /usr/share/zoneminder/www/skins/classic/views/filter.php on line 346, referer: https://zoneminder...
2.  PHP Parse error:  syntax error, unexpected '(' in /usr/share/zoneminder/www/includes/Filter.php on line 23, referer: https://zoneminder...

The first error is the same as issue #1977, just in a different place.  The second error is new.

Both of these are due to the old version of PHP used by CentOS 7.  If the resolution is to drop CentOS support, I'm actually OK with that.

**ZoneMinder Version (`zmaudit.pl -v`):**

```
# zmaudit.pl -v
1.31.43
```

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**

```
# rpm -qi zoneminder
Name        : zoneminder
Version     : 1.31.43
Release     : 1.0.20180426gitg86b2f6a12.el7.centos
[snip]
```

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**

```
# cat /etc/redhat-release
CentOS Linux release 7.5.1804 (Core)
```

Since this is a PHP error:
```
# php --version
PHP 5.4.16 (cli) (built: Apr 12 2018 19:02:01)
Copyright (c) 1997-2013 The PHP Group
Zend Engine v2.4.0, Copyright (c) 1998-2013 Zend Technologies
```

kimmeld, 2018-05-13:
Hi @knnniggett,

My install doesn't have that commit - I don't think the package containing it has been built for zmrepo yet.  It looks like zmrepo only goes as far as 86b2f6a12ed57a8e49b2427e9841c59bc6cd45f0 right now.

It looks like that line of code came in with 965b306ebc373ae15aff1e9231cdd9373f71d5e7, which matches what my install has.  Applying bbe00256418fc7c60483bc51f650f1a355839e01 results in the same error.

Changing the offending line to
```
    'sort_asc'        =>  ZM_WEB_EVENT_SORT_ORDER,
```
seems to resolve the issue, though it may be breaking other stuff that I haven't found yet.

Here is my diff for the changes to filter.php to get past the first PHP error:  [filter.txt](https://github.com/ZoneMinder/zoneminder/files/1998874/filter.txt)

Thanks!

Enjoy your vacation!

knnniggett, 2018-05-13:
Yeah, it might. I just wanted to verify the offending commit. Now that we know, we'll rewrite that line to be compatible with php 5.4 and not break anything else at the same time. Thanks!

knnniggett, 2018-05-14:
Item #1 fixed with commit 485f2929439a42903afc9b6c0624d2737d244ecf

knnniggett, 2018-05-14:
@kimmeld I've got a pr that should fix item 2, but there might be a simpler fix.
 
Can you change line 23 to the following and then report back if it works or not?
`    'sort_asc'        =>  ZM_WEB_EVENT_SORT_ORDER == 'asc' ? 'asc' : 'desc',`

Just remove the parenthesis. If this works, we'll go with the simpler fix.

knnniggett, 2018-06-05:
I was distracted a bit with a bigger zoneminder issue, but that is behind us now. I can now focus on this one. I am able to duplicate the message in the previous post and will post a fix shortly.

commit c30992be (2018-06-06): fixes #2105

A more simplified approach to fixing #2105, compared to previous attempts.
closed 2018-06-06 by knnniggett