ZoneMinder/zoneminder #2109: zmc restarting

oyhaare, 2018-05-23:
zmc keeps being restarted by zmwatch:

>zmwatch[19571]: ERR [Shared data size conflict in shared_data for monitor Monitor-1, expected 608, got 604]
> zmwatch[19571]: INF [Restarting capture daemon for Monitor-1, shared data not valid]

I suspect this might have to do with commit cc7756182699d1193d7828b21a039b9b78cc9472 and the fact that I am running on a 32bit system and/or perl vs cpp struct memory alignment/padding.

connortechnology, 2018-05-23:
c and perl should match.  I run both 32bit and 64bit and some odd things and they all work for me.  Is it latest storageareas or master snapshot?  Can't be both...

dpkg -s zoneminder

oyhaare, 2018-05-23:
I've tested with both storageareas and master.

Must be something wonky on my system then.... but what? Any ideas?
I'm running ubuntu bionic on a virtualbox vm.

I'm running ubuntu

connortechnology, 2018-05-24:
The correct value is 608.  These kinds of errors can happen when there are old libraries installed in a different location...

oyhaare, 2018-05-25:
Closing this as it seems something was messed up on my system.
Ended up reinstalling ubuntu from scratch...

closed 2018-05-25 by oyhaare

thigger, 2018-10-14:
Also getting this issue, also on Ubuntu Bionic (just upgraded from 16.04). Any idea which libraries?

cipibad, 2018-10-15:
Same issue here, fresh install 32bit Ubuntu 18.04.1 LTS, zoneminder 1.32.2-bionic1 installed from ppa:iconnor/zoneminder-1.32

cipibad, 2018-10-16:
I managed to solve the error, see: https://github.com/ZoneMinder/zoneminder/pull/2258
You can manually apply the change directly to current install modifying /usr/share/perl5/ZoneMinder/Memory.pm file.