ZoneMinder/zoneminder #2110: Issues exporting / downloading video

frankc420, 2018-05-31:
So I've upgraded to 1.31.43 and after doing so I'm now unable to export or download a video of any events.  If I try to 'download video' as Zip, it fails instantly.  If I choose Tar, it outputs a file but it is empty.

Export continuously says 'Exporting video...', so my guess is that it is instantly failing.

I'm not sure what logs to look into to find the culprit, hoping you guys can point me in the right direction.

connortechnology, 2018-05-31:
1.31.43 is pretty old.  I have fixed several issues with Export since then. Please update to 1.34.44, or more specifically the latest commit.

frankc420, 2018-06-01:
I'm using the zmrepo-testing on CentOS 7 and I'm not seeing 1.34.44.  

Sorry for my ignorance here but what is the best method of updating on CentOS if the commit isn't part of the zmrepo-testing repository?

frankc420, 2018-06-01:
I did a git checkout of the master and the version file shows 1.31.44, is your 1.34.44 a typo?  I'm assuming so.

connortechnology, 2018-06-01:
1.34.44 was a typo.

frankc420, 2018-06-02:
@knnniggett Thanks!  I actually cloned github last night so that I could get to 1.31.44 and built out using cmake instructions.  After updating the system I was unable to start zoneminder.  I had to completely remove zoneminder and re-install using zmrepo.  I'm back to 1.31.43.  I'm not sure if this is related to the issue you are troubleshooting or not, but thought I would mention it.

knnniggett, 2018-06-04:
@frankc420 I don't know if you've upgraded to the latest development rpm. We have identified the issue preventing zoneminder from starting and will push out a fix shortly. In the meantime, the workaround to get things going again is to edit /usr/lib/systemd/system/zoneminder.service and comment out the PIDFIle statement.  It seems we have a race condition at startup and systemd is looking for the pidfile before it gets created.

knnniggett, 2018-06-14:
@frankc420 This appears to be working for me on the latest master branch. Please verify after performing a yum update to the latest package, which was just synced to the mirror servers last night.
zoneminder-1.31.44-1.232.20180613gitg4b78d85b1

frankc420, 2018-06-14:
@knnniggett Thanks!  I'll keep you updated!

knight-of-ni, 2018-06-22:
If you ever have a doubt that your machine is not picking up some rpm, you can manually verify what is in the affected repo with your webrowser.  In this case the url is:
http://zmrepo.connortechnology.com/el/testing/7/x86_64/

All rpm repos work like this, with the same naming conventions for $releasever and $basearch.

As or today, if you open that url in your browser, you can see this rpm sitting there:
zoneminder-1.31.44-1.249.20180620gitg3ffd30ffb.el7.centos.x86_64.rpm

If yum isn't finding that then:
- verify [zmrepo-testing] is enabled (it's off by default)
- verify you have not excluded the zoneminder package in the repo file
- Are you using yum priorities? If yes, then you have to give zmrepo-testing a higher priority than any other repo which also has zoneminder rpms.
- issue `rpm -q zoneminder` to verify some version of zoneminder is already installed
- issue `yum clean all` then `yum update`. Watch the output and verify yum is looking in zmrepo-testing.

frankc420, 2018-06-23:
I'm not using yum-priorities but I was able to get it to install the latest build using --enablerepo=zmrepo-testing

I am now having a new odd issue, it appears my cameras are recording, but now if I click on a monitor, the image is showing broken.  If I inspect the page and grab the URL, it takes me to a dead page in the browser.  If I look in the apache log, it shows HTTP code 200 on the URL.  ZM log shows .sock file is missing.  If I look in the sock folder, the only sock file there is zmdc.sock

I've not modified the apache config at all, everything I can find on this particular issue has been related to the apache config.  So, I made a backup of my existing config and copied the /etc/zm/www/zoneminder.conf to /etc/httpd/conf.d (this is where my httpd.conf is looking) and restarted apache, no change.

I feel like I'm running in circles =\

frankc420, 2018-06-23:
[23/Jun/2018:13:12:19 -0500] "GET /cgi-bin-zm/nph-zms?scale=100&width=1280px&height=720px&mode=jpeg&maxfps=10&buffer=1000&monitor=63&auth=255c24f1ec5c18c6bb2c4dd986bdad96&connkey=442061&rand=996468 HTTP/1.1" 200 -

06/23/18 13:12:55.754139 web_php[17509].ERR [Socket /var/lib/zoneminder/sock/zms-442061s.sock does not exist.  This file is created by zms, and since it does not exist, either zms did not run, or zms exited early.  Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly.  Make sure that ZM is actually recording.  If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information.] at includes/functions.php line 2023

Odd enough, zms_m63.log shows this:  06/23/18 13:13:25.897467 zms_m63[19556].ERR-zms.cpp/203 [Unable to authenticate user]

However I've copied the source URL from ZM and dropped it into VLC and it works as expected.

juanmoura, 2018-06-25:
I found the problem in the current 1.31.44 MASTER repository.

The filter works, but when I send export it is only exporting and the file does not appear, in the log the error message of the photo of the link below appears.

[https://i.imgur.com/YWlkqNP.png](url)

connortechnology, 2018-06-25:
I think systemd now runs apache chrooted into something like
systemd-private-3c859221638e4849b0c6f3372b1582b4-apache2.service-73cwaA
So the dir will be /tmp/systemd-private-3c859221638e4849b0c6f3372b1582b4-apache2.service-73cwaA/tmp/zm
I think the final zm dir is not being created.  Maybe we need to create it.

connortechnology, 2018-06-25:
I just merged code that may fix this.

knight-of-ni, 2018-07-04:
Both @connortechnology and I have tested this on our end and have found it to be working, so I am going to close this out.

closed 2018-07-04 by knight-of-ni