ZoneMinder/zoneminder #2118: RTSP / SETUP problem without trackid causes reply with bad request 400

qpaz, 2018-06-10:
I see in wireshark with zm 1.31.1 the initial RTSP network stream SETUP packet was done with 
rtsp://hostname:port/path/trackID=X

In zm version 1.31.44 the initial RTSP SETUP packet is only
rtsp://hostname:port/path

I got a vivotek 8364 which fails in 1.31.44 on RTSP SETUP with 400 Bad Request

connortechnology, 2018-06-11:
I have no issues with my 8136 or 8166's.  

Are you using Remote or FFmpeg?

qpaz, 2018-06-12:
General
Source Type: Remote
Function: Modect
Source
Remote Protocol: RTSP
Remote Method: RTP/RTSP
Remote Host Name: <FQDN> 
Remote Host Port: 554
Remote Host Path: /live.sdp (vivotek specific)

Unitl now, this settings worked for years

knnniggett, 2018-06-12:
Using the remote source type for ip cameras has been unsupported for some time now. See the documentation: http://zoneminder.readthedocs.io/en/stable/userguide/definemonitor.html#remote

You can certainly try to use the remote source type, but we aren't going to spend any time working on it if problems occur, since better solutions exist. 

You should use the ffmpeg source type instead.

closed 2018-06-12 by knnniggett

connortechnology, 2018-06-14:
The remote rtsp was using h264 passthrough code which does not work (only works for ffmpeg).  I have pushed a fix to stop this behaviour.  New packages are building.

knnniggett, 2018-06-14:
Please be advised that we have agreed to remove the Remote->RTSP method, not in the next release, but in the release after. I will make sure and include a notice about this in the next release notes.
This will not affect the RTSP->HTTP source type.

It is in everyone's interest to switch to the ffmpeg method (or libvlc) for IP cameras as soon as possible.