ZoneMinder/zoneminder #2120: Config options in webui do nothing if set in config files [1.31.44]

MalakymR, 2018-06-11:
48f8118e5a8cf9f3de41b20a41b7b7ea9ba86a17

Updating config options in the webui do nothing when the setting exists in a .conf file. The webui also shows what I assume is the database configuration option.

If these .conf files are not to be updated from the webui (I can see why) - shouldn't the database entry take priority? Though this would mean they'd need to be null by default.

Or, disable the options in the webui from being updated to save any confusion.

knnniggett, 2018-06-11:
The last config file to be read wins. This has been tested to work as designed.

There is no database entry for the values moved to config file.... and that might be the issue. If you see for example ZM_DIR_EVENTS in Config table, then something went wrong during the update to 1.31.x.  Those values should not be there.

MalakymR, 2018-06-11:
Completely new server instance. Used git to retreive sources.

This is odd though, there is no ZM_DIR_EVENTS in the database as you say, though the settings I was testing with are the ZM_HOME_URL and ZM_HOME_CONTENT which do exist in the database.

zm_create.sql has HOME_URL etc in it, though checking source on github doesn't have it, so no idea where this has come from. Not to mention the page is loading the value in the database not the conf files - but is applying conf file settings to the page.

I just cleared the db/ folder and pulled it again and zm_create still has HOME_URL and such in it - I think this could be a caching problem, seems unlikely. Since this is not in the file here I will close this for now assuming its my problem to resolve :/

closed 2018-06-11 by MalakymR

MalakymR, 2018-06-12:
Upon pulling down sources the db/zm_create_sql.in is okay.

After cmake has finished, the db/zm_create_sql file has HOME_URL in it.

I still don't understand where this is appearing from right now.

knnniggett, 2018-06-12:
Are you trying to learn the cmake build process, or are you trying to describe some kind of problem?  Your message does not describe an actual error condition, so I'm guessing you meant the former.

ZM_HOME_URL is a variable that is defined in ConfigData.pm (which itself has a source file that gets processed by cmake called ConfigData.pm.in). See: https://github.com/ZoneMinder/zoneminder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/ConfigData.pm.in#L1493

The result is placed into zm_create.sql as a sql insert statement during build time. Scroll down to the very bottom of zm_create_sql.in and you can see the comments stating how this occurs.

MalakymR, 2018-06-13:
I am trying to figure out the process as to why this is appearing in the sql file thus showing up in the database. I missed the bottom of the .in file, admittedly I had just been greping them.

Since you said conf file settings have been removed from the database, I was looking into why this setting HOME_URL was showing up in the database even though it is set in my .conf file. I am not specifically looking at the DIR_EVENTS (I did not bring this up), that was not the issue. This may be a problem with other variables though I am just looking at HOME_URL and HOME_CONTENT right now.

The issue is still, if the variable (ZM_HOME_URL, ZM_HOME_CONTENT, unknown if others) exists in a .conf file - the project reads and uses this. The options pages do not reflect this setting and instead read from the database, and any changes to this updates the database but are not used since the conf file overrides them.

knnniggett, 2018-06-13:
The short answer is the two variables you mention do not exist in a config file on the filesystem. They should and do exist in the database Config table. What follows is the long answer....

Perhaps I was unclear. Only the system path variables, formerly found under Options -> Paths, have been moved into config files under /etc/zm/conf.d. ZM_DIR_EVENTS is just one of those variables that got moved, which is why I mentioned it.  

ZM_HOME_URL, is *not* one of the variables that got moved out of the Config table. It can still be found under Opetions -> Web. You can see that by looking at the category each of these variables has been assigned to:
https://github.com/ZoneMinder/zoneminder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/ConfigData.pm.in#L1501
https://github.com/ZoneMinder/zoneminder/blob/master/scripts/ZoneMinder/lib/ZoneMinder/ConfigData.pm.in#L1511

You should *not* see this variable under /etc/zm/conf.d. There is nothing in the code that will put this variable into a config file under that folder either.

We did not intend for the end user to override variables found in the Config table using a config file on the file system.  Although, I'm a bit surprised that doesn't work. I can take a look at that, but for now don't do that.

MalakymR, 2018-06-13:
I think the issue here is its not supposed to but it does - I've just done a scratch install and I can change a lot of these variables around and they work.

For a while I've always just assumed ZM_ variables worked in config files. I'm the one that put them there, it was the appearance in the create sql file that I couldn't figure my way back to.

Considering there is now a conf.d/ folder - it would be nice to see options still work and maybe lock them from being edited in the webui.

connortechnology, 2018-06-13:
I think what we should do is reflect the .conf value in the web ui and maybe add a note that it is being overriden in the .conf file.

knnniggett, 2018-06-13:
By the way, since you are a package maintainer, if you are trying to override a database value found in the Config dB table in your packaging scripts, a.k.a. during build time, we have a tool designed specifically to do that for you. Let me know and I can show you how to use it.

MalakymR, 2018-06-13:
I'm not trying to do this as a package default, I was just messing around with the configs and such whilst testing some bits out. I know of the script used to preconfigure these before compile time, that's clear on the front page.

Granted the original maintainer used a patch to update this info instead, so not sure if I will follow this or not. https://github.com/gentoo/gentoo/blob/master/www-misc/zoneminder/files/zoneminder-1.30.4-path_zms.patch