ZoneMinder/zoneminder #2121: zmdc.pl startup 100% CPU

cameronswinoga, 2018-06-11:
**ZoneMinder Version (`zmaudit.pl -v`):**
1.31.44
**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
Ubuntu 16.04.4

I upgraded my setup to 1.31.44 recently, and I've found that when interacting with the web controls, none of the requests to change camera settings are resolving.  I investigated a little bit, and found that the process `/usr/bin/perl -wT /usr/bin/zmdc.pl startup` is consuming 100% CPU.  In addition, there's a lot of warnings about failed linking of skins.  Past those warnings, there's no errors coming up in the logs.

MalakymR, 2018-06-11:
Have you included the cache dir options in apache?

https://github.com/ZoneMinder/zoneminder/blob/master/misc/apache.conf.in

Also permissions for apache on /var/cache/zoneminder (or whichever you end up using).

cameronswinoga, 2018-06-11:
I'm not sure what you mean by cache dir options.
```
ls -la /var/cache/zoneminder/
total 28
drwxrwx--- 5 root www-data 4096 Jun  1 12:59 .
drwxr-xr-x 1 root root     4096 Jun  1 14:58 ..
-rw-r--r-- 1 root root       29 Jun  1 12:59 configured
-rw-r--r-- 1 root root       29 Jun  1 12:59 dbcreated
drwxrwx--- 4 root www-data 4096 Jun  1 13:26 events
drwxrwx--- 2 root www-data 4096 Feb 21 08:45 images
drwxrwx--- 2 root www-data 4096 Feb 21 08:45 temp
```

MalakymR, 2018-06-11:
Your apache conf needs to be updated - above is the example. I had the same error until I updated these.

Apache wants to write do the directory which is configured.

rafique, 2018-07-14:
@cameronswinoga have you been able to solve the CPU issue? I am having the same problem: zmdc.pl takes 100% CPU after running fine couple of days.

cameronswinoga, 2018-07-16:
@rafique I ended up just killing the process whenever I restarted Zoneminder (I was running it inside of a docker container).  Past that I ended up reinstalling it from scratch into an LXC container.

vajonam, 2018-08-22:
This seems to happen when one of the cameras for me is misconfigured (while tweaking) and then zmdc tries to startup that camera after a save and takes up 100% not doing much

connortechnology, 2018-08-22:
How about some logs?

cameronswinoga, 2018-08-22:
@connortechnology I don't have any logs anymore of this happening, since I reinstalled into a seperate LXC container.

closed 2018-08-22 by cameronswinoga

vajonam, 2018-08-22:
working on it.. will post shortly, what log level is most useful? debug? trace?

vajonam, 2018-08-23:
don't have logs, but easy to repro, for me this happened when I was rotating the camera 90 degrees. I set the swapped the height and width as I was getting the settings right, and when I had them mismatched to what the stream was this happens.