ZoneMinder/zoneminder #2129: ZM_BASE_PROTOCOL is incorrect when behind a reverse proxy that connects via HTTP

mnoorenberghe, 2018-06-17:
**ZoneMinder Version (`zmaudit.pl -v`):** 1.31.44

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** 1.31.44~20180616164849-artful

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** 17.10 (Artful Aardvark)

**Relevant log lines:**
```
Blocked loading mixed active content "http://zm.example.com/zm/index.php?view=request&request=stream&connkey=1234567&command=99"
```

Visit any page that has a live stream e.g. /zm/?view=watch&mid=2

> var monitorUrl = 'http://zm.example.com';

This should be "https" in my case since the browser is loading the page over HTTPS from a reverse proxy. For JS there probably isn't a need to use the value defined by a config since a URL relative to the document the JS is running in would be more correct.

https (browser) => reverse proxy running apache => Apache with ZM over HTTP

connortechnology, 2018-06-18:
So how is the ZM Server to know it is serving https?

I think in order to support this we would have to actually have ZM_BASE_PROTOCOL be a configuration item. Perhaps we could make it be set in the Server config, instead of just a hostname, we could also include the protocol...

mnoorenberghe, 2018-06-18:
In the JS snippet I posted there is no need to use ZM_BASE_PROTOCOL at all
since it’s the front-end JS that needs the URL and a site-relative one
would work.
-- 
Sent from my phone

knnniggett, 2018-06-19:
So it looks like we are talking about these three lines: ```$ grep -r "var monitorUrl" ./
./skins/classic/views/js/zone.js.php:var monitorUrl = '<?php echo ( $monitor->Server()->Url() ) ?>';
./skins/classic/views/js/zones.js.php:var monitorUrl = '<?php echo ( $monitor->Server()->Url() ) ?>';
./skins/classic/views/js/watch.js.php:var monitorUrl = '<?php echo ( $monitor->Server()->Url() . ( ZM_MIN_STREAMING_PORT ? ':'. (ZM_MIN_STREAMING_PORT+$monitor->Id()) : '' ) ) ?>';
```
I think the reason `$monitor->Server()->Url()` returns an absolute url, rather than relative, is due to multiserver configuration. It needs to be absolute, at least in some cases, because the stream may come from another server.

knight-of-ni, 2018-06-20:
@mnoorenberghe Try modifying the Url method on your machine as shown below. Let us know if if gives the desired results: https://github.com/ZoneMinder/zoneminder/blob/master/web/includes/Server.php#L68

```
public function Url() {
    if ( ! $this->Id() ) {
        return ZM_BASE_PROTOCOL . '://'. $_SERVER['SERVER_NAME'];
    } else {
        return '';
    }
}
```

This should make the urls in question relative, unless multiserver is in use. I don't think this will break anything, but I'm still waiting to hear from @connortechnology to hear what he thinks.

connortechnology, 2018-06-20:
The code used to do exactly that.  I believe I changed it to satisfy csrf rules in a multi-port setup.

connortechnology, 2018-06-20:
I pretty much made the decision that /zm must always work, even if accessed by /.  So what about if we always append /zm when generating the url

mnoorenberghe, 2018-08-25:
> Let us know if if gives the desired results

It doesn't fix the issue. 

> I pretty much made the decision that /zm must always work, even if accessed by /. So what about if we always append /zm when generating the url

I really dislike /zm/ and tried hard to get rid of it on my setup but it seemed like that already regressed in the last release. I've accepted it for now but I'd rather we didn't require it.

Dinsmoor, 2018-08-30:
@connortechnology I'm trying to do a very similar thing as @mnoorenberghe , and I also really dislike the forced /zm/ everywhere, it makes it very very hard to use reverse proxies as well.

My example, I have a reverse proxy from this dir:
`/r/d-netbook/zm/` (dynamically created, pointing to remote servers)
which works for the front page, and any links on it. But when clicking any tabbed menus, it redirects to
`/zm/` (which is on my apache server that has no zoneminder instance, so it 404s)

I'm looking through the sources now to try to find where this behavior is, but I can't find it at this time.

mnoorenberghe, 2018-10-27:
> I also use reverse proxies and they work fine.

Do you have ZM hosted on http with the reverse proxy serving HTTPS? That is the problem that this issue is about. #2152 doesn't help since this is about the scheme/protocol, not path.

Dinsmoor, 2018-10-28:
@mnoorenberghe I am doing what you are saying is your problem.

Client > https apache2 server > proxypassed http zm backend

This is my a2cfg
Apache2 conf
<Location /r/my-remote-host/zm>
    ProxyPass http://my-remote-domain/r/my-remote-host/zm ttl=60
    ProxyPassReverse http://my-remote-domain/r/my-remote-host/zm
</Location>

That way everything translates properly, 

Although, different from your setup, I'm not using a subdomain. Protocol-wise, everything is working fine.

klausseufzger, 2018-11-14:
I had a similar problem. I use a chain of two proxy servers. One for https and one for authentifiaction rewrite, so zmNinja works (described here https://github.com/pliablepixels/zmNinja/wiki/Apache-Proxy-Authorization-with-HTTPS,-Basic-Auth,-and-ProxyPass )

zmNinja works with this configuration, but i had problems with the Montage View, because the images always used the internal Url.

I solved this by adding
`ProxyPreserveHost On`
to the proxy configurations.

gorlilla, 2018-11-15:
@knight-of-ni  

i made that change in my server.php file which resolved some of the mixed-security issues caused by the reverse proxy; however still triggered the warnings for images loaded through php-fpm. I dug around for a bit and seem to have found the other source of http urls passing through.  

i stripped the getStreamSrc in Monitors.php down to this:  i no longer have any certificate warnings when viewing in montage, cycle, monitor, etc....

`public function getStreamSrc( $args, $querySep='&amp;' ) {

    $streamSrc .= ZM_PATH_ZMS;

    $args['monitor'] = $this->{'Id'};`

mnoorenberghe, 2018-12-12:
The "Blocked loading mixed active content" error seems to have gone away since #2152.

Thanks!

closed 2018-12-12 by mnoorenberghe