ZoneMinder/zoneminder #2133: Filter doesnt work anymore in 1.31.44

qpaz, 2018-06-20:
The Filter purgewhenfull does not work anymore and today my default storage was blown up. I have no Storage under "Options --> Storage" configured. I only configured the default storage with cmake params and in conf files under /etc/zm/conf.d/10-system-paths.conf. 
When the filter purgewhenfull is executed in webfontend, results are shown. 
On bash the same filter shows 
./zmfilter.pl --filter_id=2 
06/20/2018 19:25:00.472339 zmfilter_2[15056].INF [Scanning for events using filter id '2']
06/20/2018 19:25:00.477261 zmfilter_2[15056].INF [Checking filter PurgeWhenFull delete returned 0 events ]

Kind of weird...Maybe a Bug?

1.31.44

git rev-parse HEAD
37badbfb17c374139479bac1ac806aa50de25246

NAME=Gentoo
ID=gentoo
PRETTY_NAME="Gentoo/Linux"
ANSI_COLOR="1;32"
HOME_URL="https://www.gentoo.org/"
SUPPORT_URL="https://www.gentoo.org/support/"
BUG_REPORT_URL="https://bugs.gentoo.org/"

closed 2018-06-21 by qpaz

connortechnology, 2018-06-21:
Why did you close this?

qpaz, 2018-06-23:
i think the storage overflow came up, because of many false positive events in case of rainy and stormy wheather. too many events/data were written in a short time. i had configured the purgewhenfull with 90%. 
and after the storage overflow i think this was a strange behavior of the mysql server and the database (they are using the same storage (mountpoint)). so i configured the filter to 10% for testing and after a day events were deleted correctly. i have configured now the filter to purge when disk percent is greater than or equal to 80 and migrated the database to another storage area.