ZoneMinder/zoneminder #2134: FFMpeg configured source, doesnt refresh on montage review

qpaz, 2018-06-20:
As in https://github.com/ZoneMinder/zoneminder/issues/2118 mentioned, i configured ffmpeg source type. works fine, but in "montage" the images of this cams doesnt refresh. Only one picture with timestamp is shown. in cycle watch for that source types, the images are refreshing.
I have a multiserver env with raspberry pis and source type local v4l2, they are refreshing in montage, too. Strange... In the logs i found nothing.

1.31.44

git rev-parse HEAD
37badbf

NAME=Gentoo
ID=gentoo
PRETTY_NAME="Gentoo/Linux"
ANSI_COLOR="1;32"
HOME_URL="https://www.gentoo.org/"
SUPPORT_URL="https://www.gentoo.org/support/"
BUG_REPORT_URL="https://bugs.gentoo.org/"

connortechnology, 2018-06-21:
How many cameras?  Most browsers will only connect to 6 streams at a time.  Due to the ajax calls in the background, this means 4 cameras per server.  Normally what happens is that the image never loads though, instead of just not refreshing....

We will probably need some debug logs to figure this out.

qpaz, 2018-06-23:
iam aware of the browser config mod to show all my streams. in 1.31.1 it was working. 
i will clone the current master and try to get some logs. with your build you mentioned in #2118 to support rtsp maybe i can get some hints for you...

qpaz, 2018-06-23:
ok, i figured out the problem here too... there were some odd files in var/tmp/zm and the storage was full ....after deleting the files the montage is running fine .... i checked if zm is deleting the files there, after closing montage... looks all pretty damn good ... sry for wasting your time connor, ty

closed 2018-06-23 by qpaz