ZoneMinder/zoneminder #2138: Integrate OpenCV's image detection into zm_monitor.cpp as an alternate option

pliablepixels, 2018-07-02:
OpenCV offers some "reasonable" methods to detect people. They are not as accurate as CNNs and use HOGs but are of orders of magnitude faster. The code is straight forward, for example, [this](https://github.com/opencv/opencv/blob/master/samples/cpp/peopledetect.cpp). OpenCV code is very portable, and in C++, and therefore integration into ZM is not hard. (I don't feel like investigating - someone who knows zm_monitor.cpp and how ZM handles image buffers should be able to do it much faster than me)

There are multiple HOGs to choose from (different XML files)

The request: 
a) Integrate OpenCV with ZM 
b) Select "Person Detection" as a scheme as a detection algorithm as an alternate to "Pixels/Filter/Blob" and basically replicate the functionality provided in the link above.
c) Ability to specify a "frame interval" -> which will basically decide how often this logic will be used. In other words, even if I select the monitor FPS as say 5, I should be able to select "2" here, which will mean detection to be applied on the 1st, 3rd, 5th frames - for more speed control without affecting live fps
d) Draw a bounding box around the person detected in frame (Again, already part of the sample code)

I'll be putting in a $100 bounty for this.

pliablepixels, 2018-07-02:
bounty added

connortechnology, 2018-08-11:
So I have merge the CVisionLabs work into my opencv branch.  However the plugin code is for an ancient version of opencv.... so I think I'll be updating that to version 3.

pliablepixels, 2018-08-11:
do you have a branch I can track/test?

connortechnology, 2018-08-11:
my opencv branch

TheSwert, 2018-08-25:
Since version 3.4.2 of OpenCV, it looks like it now supports the YOLOv3 DNN detector as well. Below link has some C++ and python examples.
https://www.learnopencv.com/deep-learning-based-object-detection-using-yolov3-with-opencv-python-c/

pliablepixels, 2018-10-08:
I'm going to revise some of the asks based on my experience [integrating both OpenCV approaches](https://github.com/pliablepixels/zmeventserver/tree/master/hook_example) in the ZM event notification server:

I originally said:

> b) Select "Person Detection" as a scheme as a detection algorithm as an alternate to "Pixels/Filter/Blob" and basically replicate the functionality provided in the link above.
> c) Ability to specify [...]

I don't think this is practical.

Based on my experience,
a) Many (if not most) users of ZM are not running a GPU
b) OpenCV SVM HOG is very fast but not quite accurate. It keeps identifying my water heater as a "person"
c) OpenCV DNN with Yolov3 (80 labels) is quite accurate but without a GPU takes 10x the time as HOG (On an Intel Xeon 3.16GHz 4Core machine with no GPU HOG takes 0.25s and YoloV3 takes 2.4s)

To make this practically useful, I'd like to suggest that "Person detection" is implemented as a "post analysis" option to "motion" not as an alternate.

In other words, ZM first detect motion, using whichever means and _then_ runs person detection before an alarm triggers. This is the approach I am following, and even with a 2s analysis, its quite acceptable because it only kicks in _after_ ZM decides an alarm has occurred.

closed 2018-12-16 by stale[bot]

3lk0k0, 2018-12-24:
anyone know if these if going to do ?

connortechnology, 2018-12-24:
I started on it... but there is lots more to do.

jysaloma, 2019-02-10:
There is also some work done in this project for face recognition https://github.com/goruck/smart-zoneminder by goruck

cleverinx, 2019-12-01:
Is this still in progress?

connortechnology, 2020-06-17:
bountysource will take this money on July 1st.  Either contact bountysource to redirect your funds, or alternatively, I can close this out, collect the bounty, and I will hold the money until this issue is properly closed.

pliablepixels, 2020-06-17:
I’d prefer you just collect for the money I added.

Rayn0r, 2020-11-10:
@connortechnology 
Is implementing OpenCV into Zoneminder still on the list of things to do?

connortechnology, 2020-11-10:
Yes

connortechnology, 2023-09-05:
Yes it's still open. I'm not sure it's a good idea, as I think AI should be happening in it's own process.