ZoneMinder/zoneminder #2140: 1.30.[2,3,4]: Crud is missing, incomplere source tarballs

onlyjob, 2018-07-02:
Submodule _Crud_ is missing from source tarballs of last threee releases: directory `web/api/app/Plugin/Crud` is empty hence API is broken...

knight-of-ni, 2018-07-03:
That's correct. The source tarballs you are referring to are autogenerated by git and do not contain submodules. This is not a problem for any other distro's build system out there. Please explain why this is an issue for Debian.

onlyjob, 2018-07-03:
So Crud is no longer required?
If it is required then source tarball is not comprehensive (incomplete) which is a problem.

This is an issue for Debian because everything is built from source tarballs in offline build environment in order to build _reproducibly_.  

Regardless, incomplete tarball is a legitimate issue, is it not?

My co-maintainer (had to?) [patch away](https://salsa.debian.org/debian/zoneminder/commit/2a3db2ae) submodules in order to fix FTBFS on build. I suspect some other distros may be doing the same.

Download on build-time is a bad practice and it is not a solution to incomplete tarball.
It is a privacy breach and a potential security vulnerability.

onlyjob, 2018-07-03:
So how Arch or Fedora generate comprehensive tarball? You are using Fedora, right? I imagine you might know how it is done by Fedora...

knight-of-ni, 2018-07-03:
Yeah, I am the Fedora package manager for ZoneMinder and several of its dependencies.

You can view the latest zoneminder rpmspecfile here, showing multiple source tarballs:
https://github.com/ZoneMinder/zoneminder/blob/master/distros/redhat/zoneminder.spec#L41

Here is the official Fedora packaging documentation, which even reminds the packager that submodules are *not* part of the original source tarball: https://fedoraproject.org/wiki/Packaging:SourceURL#Git_Submodules

If goes on to describe how to handle it.

Other then providing some high level steps, I'm not sure how useful this is going to be for you.
Let me do some searching. I'll see what I can find that is more Dedian specific.

NOTE: There are some additional modifications you need to make to get Debian to work properly on the latest master, soon-to-be 1.31.2.  A summary can be found here:
https://github.com/ZoneMinder/zoneminder/issues/2078#issuecomment-395094624

Tor now, this is just something to be aware of. Let's deal with that later.

onlyjob, 2018-07-03:
Thank you very much for hints. I've implemented proof of concept of multi upstream tarball (MUT) for Zoneminder and I'm hoping to upload it with Crud properly soon... I've found exporting individual submodules as tarballs to be unexpectedly difficult but doable...

knight-of-ni, 2018-07-03:
Check out the following the blog. I believe this is the approach you want to use:
https://raphaelhertzog.com/2010/09/07/how-to-use-multiple-upstream-tarballs-in-debian-source-packages/

It steps you through adding an additional source tarball to the spamassassin package. Indeed, this appears to have been done already in the current Debian repos. I fired up a vm running stretch and then checked out the spamassassin source. The additional tarball mentioned in the referenced blog shows up in the source files.  

So this appears to be the procedure:
- Grab the Crud tarball. I believe this is the url you want to use: https://github.com/ZoneMinder/crud/archive/v3.0.10.tar.gz
- Re-archive it into zoneminder_1.30.4.orig-crud.tar.xz
- Unpack it into a folder called curd, in the root of source files
- Modify rules to install crud into api/app/Plugin folder

There may be hidden git files in the Plugin folder and you may have to remove them first. I do this in the rpm package.

Let me know if this is helpful or not. I can continue to search for other examples if you want.

Something else I should mention. The versions of CakePHP and Crud, which make up ZoneMinder 1.30.4, are not compatible with PHP 7. These libraries have been upgraded in the latest master branch, but not backported to 1.30.4. 

Once we get ZoneMinder set up properly for multi-source, I'd rather we look at getting ZoneMinder 1.32.0 into Debian. However, if you'd rather backport the changes to 1.30.4 then I am willing to work with you on that. Let me know.

Oh, and my best guess for 1.32.0 release is by end of summer. Possible sooner, if I can get help resolving the last remaining issues.

onlyjob, 2018-07-03:
Thank you, thank you. I've seen multi upstream tarball (MUT) sources and used the technique already. I was struggling to automate making all the tarballs using our standard tools and `debian/watch` and to some extent succeeded...

I've just uploaded updated "cakephp" 2.10.11 package to take care of PHP 7.2 compatibility.
Zoneminder in Debian was already using external "cakephp" and I've improved that in pending upload. Much appreciate that you pointed out to PHP-7.2 compatible Crud release. Thanks.

I'm looking forward to prepare 1.32.0 for upload. In the meantime I might have to cherry pick some fixes for 1.30.4...

Thanks again and good luck with releasing 1.32.0. It will be a big release as it seems...

onlyjob, 2018-07-04:
OK, I've [stabilised](http://metadata.ftp-master.debian.org/changelogs/main/z/zoneminder/unstable_changelog) 1.30.4 [in Debian](https://packages.qa.debian.org/z/zoneminder.html), with recent patched Crud from submodule. All good.

I'm closing this tickes as Zoneminder project is not publishing comprehensive source releases apart from tagging repository. Fair enough, problem is solved downstream although it could have been much easier to work with complete source tarball...

@knight-of-ni, you did not have to lock #2142. Not from me anyway. I've found the root cause and I would have closed it myself with a note how to fix the issue for anyone else who encounters the same problem. Unfortunately I can not add my comments to the ticket...

I'll see if I can spare some time for testing of _master_ in its pre-release state. From a quick glance I've noticed some problems with Javascripts dependency hygenie like pre-built/minified source-less files that we should replace for DFSG compliance, to make sure that truly Open Source distribution ships all sources. :) I might be able to give a hand with that. Thanks.

closed 2018-07-04 by onlyjob