ZoneMinder/zoneminder #2141: H.264 Storage with Maximum FPS values set causes crash

poldon, 2018-07-03:
I believe I have found a bug when enabling the X264 Encode and H.264 Camera Passthrough options.  If the Alarm Maximum FPS and/or other Maximum FPS settings are set, the capture process repeatedly crashes and restarts.  I saw the following log entry multiple times each time it crashed:

2018-06-28             08:45:36.338082 | zmc_m2 |   | 25680 | WAR | Unable to receive             frame 231: Resource temporarily unavailable, continuing | zm_ffmpeg_camera.cpp | 896

Let me know if you need more info.  Once I removed the Maximum FPS settings, everything cleared up.

**ZoneMinder Version (`zmaudit.pl -v`):** 1.31.44~20180625155445-bionic

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):**  I'm using the Zoneminder-Master PPA.

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** Ubuntu 18.04

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:**  Generic ONVIF Camera, 24 FPS capture using RTSP

**Relevant log lines:**
```
2018-06-28             08:45:36.338082 | zmc_m2 |   | 25680 | WAR | Unable to receive             frame 231: Resource temporarily unavailable, continuing | zm_ffmpeg_camera.cpp | 896

```

knight-of-ni, 2018-07-03:
What value did you place in the max fps field? More precisely, was it higher or lower than the frame rate coming from the camera?

knight-of-ni, 2018-07-03:
ok, thank you. That's good info.

connortechnology, 2018-07-04:
Uh Oh.  This line is in the experimental hwaccel decoding code.  Which is known to not work.  It indicates that the hardware accelerated decode failed.

connortechnology, 2018-07-04:
We are going to need more detailed debug logs.