ZoneMinder/zoneminder #2142: 1.30.4: no images in "Show Timeline"

onlyjob, 2018-07-03:
Ever since upgrade to 1.30.4, _Show Timeline_ have no images.

This regression is quite interesting: old events are rendering correctly in the timeline but new events refer to missing images (as per web server logs) despite Events being playable...

knight-of-ni, 2018-07-03:
The timeline is known to work fine, even on Debian.
I'm closing this as you are reporting this against 1.30.4, and we are going to release 1.30.2 soon. If you want someone to spend time on this, then test against the latest master branch.

closed 2018-07-03 by knight-of-ni

onlyjob, 2018-07-03:
Please don't be so hasty closing bugs. It looks like legitimate issue. So far I've tried to backport various fixes unsuccessfully. Maybe something else is causing this as timeline references to images that do not exist in "events".

I'll see if I can test "master" but I would expect most recent release to be somewhat supported.

In the meantime I would like to stabilise 1.30.4...

When is new release is planed to be tagged?

Besides "even on Debian" is not nice... :(

knight-of-ni, 2018-07-03:
I agree that there certainly is some kind of issue somewhere in there. However, you ignored the new issue instructions and just posted "hey, I got a problem" with no data to speak of, for a version of zoneminder we are not working on. There are better places to get help on this issue.

If you search our github forum, you'll find a relatively recent thread from a Debian user who reported a similar problem and then realized one of three places to set the timezone was off. That is something to look at.

We also have an extensive user forum exactly for questions that are short on data. I will be glad to help you there. I'm also on IRC. Our github forum is the only forum where restrictions are placed on the content. Please understand this isn't personal. This is not against you nor Debian. 

When creating issues, please fill out the requested information and include real data. This is for my sanity. I don't get a lot of help managing this forum, so help me help you, and let's keep the content objective and always include the actual error messages. Otherwise our forum just overflows with things that won't ever get looked at... and then nothing gets worked on at all because no one can tell the difference between the junk and the real stuff.