ZoneMinder/zoneminder #2143: Montage review, time selector

armonica17, 2018-07-05:
Fedora 27, built fresh using zoneminder instructions. Ongoing problem.
I had zoneminder running on a DL380 for years as a VM. I could easily take a slider, pull up say the 8 hour slider and quickly review what happened. The old  RHEL version became dated and I simply had to rebuild. The new interface looks better - great job! Problem is I'm not able to use that slider while looking at the four monitors like I used to. In fact the slider doesn't appear at all. If I pull a camera up individually it would let me review that one and even jump from camera to camera in respect to time, which is nice.
Cameras are set up for modect, 10 FPS, 1080. I tried setting things until I broke it. I deleted the VM and started all over reading the manual on how to do this. I have a feeling something isn't set that needs to be however I'm not finding it via google or poking through tickets. User error?

Thanks,
Rob

knight-of-ni, 2018-07-05:
It's always good to see other Fedora users participating in the debug process.

Once you select a range: 1 hour, 8 hour, etc, click inside the horizontal header bar that appears. A thin, vertical bar should appear inside the horizontal header. This is your slider. Click and drag appropriately.

I'm testing this right now in my own Fedora 28 vm, and it is working for me. The screen does appear rather crowded, so its possible the html could is overlap or otherwise not display properly if the screen resolution is not high enough.

In any case, if the bar does not appear, try switching different themes to see if it is specific to just one theme. 

Since the bar uses JS, verify the zoneminder logs are free of js errors. Open up your browser console by "inspecting" the page source. Look in the browser debug console to see if there are any js errors happening.

As a side note, I just fixed a problem where the development rpm was deleting all events after every reboot. I also found and fixed an issues where the build was not setting certain default values properly during the rpm build for new databases. I know you said you did a fresh build, but make sure your build is as recent as 14b83ac.  You can tell by comparing "git rev-parse --short HEAD" to the list here: https://github.com/ZoneMinder/zoneminder/commits/master

armonica17, 2018-07-06:
Hi, I think I may have mislead you. I'm not using the git copy. I have the rpm version - zoneminder-1.31.44-1.269.20180703gitg14b83ac36.fc27.x86_64, which looks like the one you want.
I have errors. Attached - 
[image]

Hope this helps. I changed from classic to base for the css. I don't seem to have a flat theme.

Thx,
Rob

knight-of-ni, 2018-07-06:
Hey look at that, logger.js is no longer around. Should be easy to fix. I don't think that will effect montage review, but it should clean up the error log.

To change themes go to Display -> CSS. Today, your choices are base, classic, and dark. There is no flat. 

Can you confirm you meant to close this issue out? Your last message says the slider is not there at all.

armonica17, 2018-07-06:
Hi. I certainly didn't mean to close it out. I must have hit the wrong button.

I wanted to make sure I really did change the themes. I did. Classic to base, no change. For the montage review I'm not seeing the slider. 1 hour or 8 hour.

knight-of-ni, 2018-07-06:
Can you provide a screenshot of what you do see? I can't duplicate this, and I'm running in nearly the same environment as yours.

armonica17, 2018-07-06:
Attached is a gimp SS. I pulled up 8 hour, clicked right in the middle of that slider and I saw the logger.js error, then TypeError: stack is undefined index.php:83:7.
[image]

knight-of-ni, 2018-07-13:
@armonica17 The good news is that we did discover some issues with Montage Review and the fixes are contained in the latest F27 RPM in zmrepo testing. You should dnf update to get those.

The bad news is that we still have not been able to duplicate the error message you are seeing so I'm not confident this latest will fix the issue for you.

If the issue still persists after updating, we need to take a look at how your particular system is configured to hopefully find a clue as to what is causing this.  By that I mean reporting any changes you can think of, no matter how small, that were made under Options or to the config files under /etc/zm. things like enabling authentication, adding or changing storage areas, additions to the Servers tab, etc.

In addition, it looks like you have four cameras from the screenshot. In order to rule out something with one of the monitor configs, try disabling one camera at time to see if the error goes away.

armonica17, 2018-07-14:
I ended up creating a new VM, F28 and loading a camera. Ran the usual tests - firefox, chrome, etc. I found that it'll work under Chrome, won't work in firefox. At least not on this machine. I fired up my laptop that has F28 loaded and it works fine with firefox. I updated the F27 version with the 4 cameras and same thing. Firefox doesn't work, chrome does, on the laptop it works fine. It didn't used to work with the laptop's firefox. I even tried a Windows box with IE at one point. I have a feeling your update fixed it.

Thanks.

closed 2018-07-14 by armonica17