ZoneMinder/zoneminder #2146: Excessive warnings about failing attempts to find `arp` utility

onlyjob, 2018-07-06:
https://github.com/ZoneMinder/zoneminder/blob/master/web/skins/classic/views/monitorprobe.php#L300

I get this warning all the time: `Unable to determine path for arp command, type -p arp returned '127'`

Because location of `arp` command can not be found like this:

https://github.com/ZoneMinder/zoneminder/blob/master/web/skins/classic/views/monitorprobe.php#L298

There are many problems with that approach.

~~~~
$ sh -c 'type -p arp'; echo $?
-p: not found
arp: not found
127
~~~~

That's because default non-interactive shell on Debian is _dash_ which have no `-p` option.

Let's try again:

~~~~
$ sh -c 'type arp'; echo $?
arp: not found
127

$ bash -c 'type arp'; echo $?
bash: line 0: type: arp: not found
1
~~~~

Neither _dash_ nor _bash_ could find `arp` because it is not in user's PATH.
On Debian, `arp` is in `/usr/sbin/arp` (hence `sudo sh -c 'type arp'` could find it).

But back to the original problem: there is no point to warn about failing attempts to locate `arp` utility if last fallback to hard-coded path works.

I suggest to remove all the warnings and replace them with only one, just [before the following line](https://github.com/ZoneMinder/zoneminder/blob/master/web/skins/classic/views/monitorprobe.php#L317) to warn only if `arp` hasn't been found after all attempts.

_Only one warning is enough._

Current situation is misleading because false warning led me to believe that monitor probe is failing because of missing `arp` utility while it was actually entirely unrelated problem: #2144.

knight-of-ni, 2018-07-07:
For entertainment purposes, you should search our github forum to see how much time we have spent revisiting this, trying to find a single solution that works on debian, redhat, and freebsd distros.

I don't agree that only one warning is enough because it does not tell you exactly what part(s) failed, which is necessary for troubleshooting. However, I think we can come to compromise that replaces the warnings with debug lines, and then place a single warning at the end that just says something like "probe failed, enable debug for details". I will look into this further when time allows.

onlyjob, 2018-07-07:
I can imagine how much time it would consume...

One warning (only) when all attempts fail + debug per attempt perhaps is the best.
It would dramatically reduce unhelpful noise in logs.

Thanks.

connortechnology, 2018-07-10:
I suspect that in debian apache/php when it exec's does not use bash, probably /bin/sh instead.  Under /bin/sh type -p arp returns 

-p: not found
arp is /usr/sbin/arp

onlyjob, 2018-07-11:
Indeed. :)
Just as I've said, 

> That's because default non-interactive shell on Debian is _dash_ which have no -p option.

knight-of-ni, 2018-07-11:
So here's a thought. During the build, cmake uses find_program to search the build environment for the ffmpeg binary. If it finds it, it populates the full path to the binary in ZM_PATH_FFMPEG via make macro. The end user is free to change the value at any time after ZoneMinder has been installed. 

We can do the same thing with ZM_PATH_ARP and then just remove all this inefficient php code that tries to find the right tool to use. This would require package managers to include the arp tool as a package dependency.

connortechnology, 2018-07-11:
That's pretty much exactly my thought as well.

PR #2155 (2018-07-11): Arp tool
Fixes #2146 

Probe works w/o throwing errors on my test vm, but I don't have any hardware the probe will detect.
closed 2018-07-11 by connortechnology