ZoneMinder/zoneminder #2147: Outdated csrf-magic library

onlyjob, 2018-07-06:
_csrf-magic_ appears to be unmaintained/abandoned upstream. Yet, despite being untouched upstream in years we don't even use the latest version:

https://github.com/ZoneMinder/zoneminder/blob/master/web/includes/csrf/csrf-magic.php#L134

Here I've found v1.0.5: http://repo.or.cz/csrf-magic.git whihch can be cloned as follows

    git clone http://repo.or.cz/csrf-magic.git

Please update it.

SteveGilvarry, 2018-07-06:
Happy to compare but we are using this version I believe. More recent
updates and activity from author, but no version update.
https://github.com/ezyang/csrf-magic
Will look closer over weekend.

onlyjob, 2018-07-06:
Thanks. Let's clarify with upstream: https://github.com/ezyang/csrf-magic/issues/14

closed 2018-09-11 by stale[bot]

onlyjob, 2018-09-11:
When legitimate bugs are closed like this I feel _very_ discouraged from reporting more and I'm sure I'm not the only one. Closing tickets (without action) does not fix problems or make them go away.
This is just an _incompetent_ bugs handling. :(

netniV, 2020-02-23:
Just for reference, we use stale bot on our repositories too.  If someone had responded with 7 days of the warning, it would have stayed open.  But I see no point in leaving issues open that aren't active or being worked on.

I'm only here because I just posted an update to the CSRF stuff on my personal fork of it, to add some new features and allow us to move away from sha1 at some point in the future, and was looking at the source open issue list.