ZoneMinder/zoneminder #2148: "Add New Monitor" dialog fails to apply camera properties detected by "Probe"

onlyjob, 2018-07-06:
After fixing #2144, I've found another problem: detected network camera information is not applied in "Add New Monitor" dialog.

Here is what happenning:

  Add New Monitor --> Probe (camera chosen, click "Save")
  Probe dialog closes, "New Monitor" dialog reloads without changing any parameters from detected camera.

From what I can tell, "New Monitor" dialog reloads by POST from "Probe" dialog, with correct camera data in `__csrf_magic` yet it doesn't apply to form somehow...

connortechnology, 2018-07-10:
The issue was that we used to base64_encode the probe data, but no longer do that, so the base64_decode was creating junk.

closed 2018-07-10 by connortechnology

onlyjob, 2018-07-11:
d05ad12 fixes the problem. Thank you.

commit d05ad12b (2018-07-10): We no longer base64_encode the probe data, we just use htmlspecialchars, so don't base64_decode it.  Fixes #2148