ZoneMinder/zoneminder #2149: ship only one version of jQuery

onlyjob, 2018-07-07:
Let's ship only one version of jQuery please.

Currently there are two:

  * 1.11.3: `web/skins/classic/js/jquery-1.11.3.js`
  * 1.12.4: `web/skins/classic/js/jquery-ui-1.12.1/external/jquery/jquery.js`

onlyjob, 2018-07-11:
I've just found aother one in `web/skins/classic/js/chosen`, version 1.8.2:

https://github.com/ZoneMinder/zoneminder/blob/master/web/skins/classic/js/chosen/chosen.jquery.js
https://github.com/ZoneMinder/zoneminder/blob/master/web/skins/classic/js/chosen/chosen.jquery.min.js

onlyjob, 2018-07-12:
Could someone re-open this bug please before we all forget? Thanks.

connortechnology, 2018-07-12:
Um, that is not jquery, that is chosen, using the jquery api.  We can get rid of chosen.proto.* though.

onlyjob, 2018-07-14:
Ah, right, my mistake... All good then. Thanks.

commit 77fbfcce (2018-07-10): fix #2149
closed 2018-07-14 by onlyjob