ZoneMinder/zoneminder #2156: Add login/logout via API

pliablepixels, 2018-07-12:
This PR will build on top of @connortechnology 's session fixes via API and introduce login/logout APIs.
I plan to keep the `AppController` `user=&pass=` intact - doing it via any API is also a convenient option.

pliablepixels, 2018-07-13:
I can confirm the same problem with `login.php?user=admin&pass=whatever` - does not work with curl, but works with postman and zmN

```
curl -X POST \
  'https:/myserver/zm/api/host/login.json?user=admin&pass=xxxx' \
  -H 'cache-control: no-cache' \
  -H 'content-type: application/x-www-form-urlencoded'
```

connortechnology, 2018-07-13:
Um, this isn't how you do post with curl.  When you stick the data in the url like that it is still a get.  There is another parameter -d for passing the post data.

pliablepixels, 2018-07-13:
Not sure what you are referring to. I am not putting data in the url for post.

connortechnology, 2018-07-13:
uh I don't know where it has gone but I had an email where you posted a curl command line that didn't work and you had ?user=&pass= in the url  passed to curl... I swear...

pliablepixels, 2018-07-13:
I posted two examples in the PR. One with POST and -d (new way) and one with GET and url params (your original  commit). Both exhibited the same curl issue.  It was posted here https://github.com/ZoneMinder/zoneminder/pull/2157

closed 2018-07-16 by pliablepixels