ZoneMinder/zoneminder #2159: Saving Monitor fails - Invalid Date Format

kstst, 2018-07-16:
Fresh Install ZoneMinder 1.30.4 
Fresh Fedora 28 workstation 4.17.5-200.fc28.x86_64.

Error
SQL-ERR 'SQLSTATE[22007]: Invalid datetime format: when saving monitor.

Documentation Needs to be added to README.Fedora

Issue with Mariadb
SQL_MODE has been changed; in particular, NOT NULL fields with no default will no longer fall back to a dummy value for inserts which do not specify a value for that field.

Fix
Add Following to /etc/my.cnf
[mysqld]
sql-mode="NO_ZERO_IN_DATE,NO_ZERO_DATE"

Restart Mariadb

Credit
 http://mamchenkov.net/wordpress/2017/11/30/change-sql-mode-for-mariadb-in-fedora-27/

 Notes
mysql -uzmuser -p -e  "SELECT @@SQL_MODE, @@GLOBAL.SQL_MODE;"
Before:
 STRICT_TRANS_TABLES,ERROR_FOR_DIVISION_BY_ZERO,NO_AUTO_CREATE_USER,NO_ENGINE_SUBSTITUTION 
After:
 NO_ZERO_IN_DATE,NO_ZERO_DATE | NO_ZERO_IN_DATE,NO_ZERO_DATE

knight-of-ni, 2018-07-16:
I'm fairly sure we fixed these kinds of sql errors in the master branch by modifying our sql inserts to explicitly insert the default value rather than let sql implicitly do it. Please test against the latest 1.31.44 development by either building your own rpm (instructions are on readthedocs) or grabbing an rpm from zmrepo.

kstst, 2018-07-17:
How can we let people know that the recommended method to install will end up with a non-functional system?
Any idea when the Fedora 28 repo going to be updated? 

Readme.md
Install from a Package Repository
This is the recommended method to install ZoneMinder onto your system. ZoneMinder packages are maintained for the following distros: Fedora 28 is listed.

knight-of-ni, 2018-07-17:
> How can we let people know that the recommended method to install will end up with a non-functional system?
Simple. ZoneMinder 1.32.0 will be released very soon. They can also read this forum.

> Any idea when the Fedora 28 repo going to be updated?
I'm not sure what you mean. Which repo are you referring to?
RPMFusion hosts ZoneMinder releases while zmrepo hosts the latest ZoneMinder development packages. This is in our documentation.

BTW: ZoneMinder 1.30.4 is incompatible with php 7 which ships with f28. The API will not work. That has also been fixed and will be part of the next release.

knight-of-ni, 2018-07-17:
If you want the next release to come sooner, then help us out by testing the issue you opened against the latest package.

closed 2018-08-02 by kstst