ZoneMinder/zoneminder #2160: zmvideo.pl -- "sh: -y: invalid option"

thoraxe, 2018-07-17:
**ZoneMinder Version (`zmaudit.pl -v`):**
1.30.4

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
CentOS Linux release 7.5.1804 (Core)

When trying to generate a video via the web:

**Relevant log lines:**
```
==> zoneminder/zmvideo.log <==
07/16/18 13:10:14.787423 zmvideo[1182].INF [Creating video file Event-3937-r1-s1.avi for event 3937]

==> httpd/error_log <==
sh: -y: invalid option
Usage:  sh [GNU long option] [option] ...
        sh [GNU long option] [option] script-file ...
GNU long options:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --protected
        --rcfile
        --rpm-requires
        --restricted
        --verbose
        --version
Shell options:
        -irsD or -c command or -O shopt_option          (invocation only)
        -abefhkmnptuvxBCHP or -o option
        
....

==> zoneminder/zmvideo.log <==
07/16/18 13:18:36.410832 zmvideo[588].INF [Creating video file Event-3937-r1-s1.avi for event 3937]
07/16/18 13:18:36.558057 zmvideo[588].ERR [Unable to generate video, check /usr/share/zoneminder/www/events/1/18/07/15/02/50/00/ffmpeg.log for details]

```

Also happens directly from CLI:

```
[root@416b25c0d2d4 /]# /usr/bin/zmvideo.pl -e 3937 --format mpg --rate=100 --scale=1 --overwrite
sh: -y: invalid option
Usage:  sh [GNU long option] [option] ...
        sh [GNU long option] [option] script-file ...
GNU long options:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --protected
        --rcfile
        --rpm-requires
        --restricted
        --verbose
        --version
Shell options:
        -irsD or -c command or -O shopt_option          (invocation only)
        -abefhkmnptuvxBCHP or -o option
```

Perl debug takes you here:

```
main::(/usr/bin/zmvideo.pl:222):                my $video_file = $Event->GenerateVideo( $rate, $fps, $scale,  $size, $overwrite, $format );
```

Looks like something weird going on in GenerateVideo but this is where my debugging skill ends. It's possibly a weird issue with RHEL/CentOS vs. other OSes.

thoraxe, 2018-07-17:
Note this is not related to #1146 despite having the same log entries. I even tried that `qx` fix with the slash but it didn't help -- not that I expected it to, as the Perl debugger kicked me out in a different area.

knight-of-ni, 2018-07-17:
Your latest reply came up as I was typing.
ZM_PATH_FFMPEG is supposed to be set during packaging but it was not because the ffmpeg binary was not being brought in at build time. This was previously fixed in commit 14b83ac36b4bc64070fe36e116f8c6b6a13c9219

closed 2018-07-17 by knight-of-ni

knight-of-ni, 2018-07-17:
@thoraxe I just wanted to say thanks for taking the time for describing the issue so well, using precise objective language. Not everyone knows how to do that. It allows us to quickly arrive at a conclusion.

For completeness I also added commit 455204135da89df083a67851b3d48b1a89b2509d to fail with a more specific message if the ffmpeg binary is not set.

thoraxe, 2018-07-17:
@knight-of-ni as a software company employee (Red Hat) who works with engineers and product managers all day, I know how to write bugs and product requirements :) I'm glad the detail was useful.