ZoneMinder/zoneminder #2169: Install difficulty, Linux Mint 18.3: "Insecure directory in $ENV{PATH}"

BassmanBiff, 2018-07-25:
Forgive me if I'm missing anything important, I'm definitely a noob in this stuff!

I'm trying to follow the directions [here](http://zoneminder.readthedocs.io/en/latest/installationguide/ubuntu.html#easy-way-ubuntu-16-04), on Linux Mint 18.3, but on "service zoneminder start" (step 7), I get the following:

> "Job for zoneminder.service failed because the control process exited with error code. See "systemctl status zoneminder.service" and "journalctl -xe" for details."

The systemctl command gives the following:

> ● zoneminder.service - ZoneMinder CCTV recording and surveillance system
>    Loaded: loaded (/lib/systemd/system/zoneminder.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Wed 2018-07-25 16:17:17 CDT; 8s ago
>   Process: 23773 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=2)
> 
> Jul 25 16:17:17 carme systemd[1]: Starting ZoneMinder CCTV recording and surveillance system...
> Jul 25 16:17:17 carme zmpkg.pl[23773]: Insecure directory in $ENV{PATH} while running with -T switch at /usr/share/perl5/ZoneMinder/Memory.pm line 138.
> Jul 25 16:17:17 carme zmpkg.pl[23773]: Compilation failed in require at /usr/share/perl5/ZoneMinder.pm line 37.
> Jul 25 16:17:17 carme zmpkg.pl[23773]: BEGIN failed--compilation aborted at /usr/share/perl5/ZoneMinder.pm line 37.
> Jul 25 16:17:17 carme zmpkg.pl[23773]: Compilation failed in require at /usr/bin/zmpkg.pl line 48.
> Jul 25 16:17:17 carme zmpkg.pl[23773]: BEGIN failed--compilation aborted at /usr/bin/zmpkg.pl line 48.
> Jul 25 16:17:17 carme systemd[1]: zoneminder.service: Control process exited, code=exited status=2
> Jul 25 16:17:17 carme systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.
> Jul 25 16:17:17 carme systemd[1]: zoneminder.service: Unit entered failed state.
> Jul 25 16:17:17 carme systemd[1]: zoneminder.service: Failed with result 'exit-code'.

I took a look at zmpkg.pl, and tried replacing the architecture detection chunk with `$arch = 64;` Whether that's correct or not, it let things continue, and found more "insecure directories" in other files. After I was done "fixing" all of them (probably breaking them in the process), it still fails, but the systemctl output has no information:

> ● zoneminder.service - ZoneMinder CCTV recording and surveillance system
>    Loaded: loaded (/lib/systemd/system/zoneminder.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Wed 2018-07-25 16:04:01 CDT; 1s ago
>   Process: 1523 ExecStart=/usr/bin/zmpkg.pl start (code=exited, status=1/FAILURE)
> 
> Jul 25 16:04:01 carme systemd[1]: Starting ZoneMinder CCTV recording and surveillance system...
> Jul 25 16:04:01 carme zmpkg[1523]: INF [Sanity checking States table...]
> Jul 25 16:04:01 carme zmpkg[1523]: INF [Command: start]
> Jul 25 16:04:01 carme systemd[1]: zoneminder.service: Control process exited, code=exited status=1
> Jul 25 16:04:01 carme systemd[1]: Failed to start ZoneMinder CCTV recording and surveillance system.
> Jul 25 16:04:01 carme systemd[1]: zoneminder.service: Unit entered failed state.
> Jul 25 16:04:01 carme systemd[1]: zoneminder.service: Failed with result 'exit-code'.

I'm guessing that my "fixes" just broke stuff, but I was hoping to at least get some info from it. What's the *right* way to fix this? As a side note, I notice that the guide I'm following mentions php5 but the guide ends up installing php7, if that's an issue.

Thank you!

connortechnology, 2018-07-27:
Can you look into something for me?  do 

ls -l /bin

and 

ls -l /bin/uname

BassmanBiff, 2018-07-30:
Done: https://pastebin.com/BD3VxyQx

closed 2018-10-08 by stale[bot]