ZoneMinder/zoneminder #2178: ZoneMinder ONVIF errors on add (hikvision)

Cramac, 2018-08-12:
same problem.
when trying to add a camera via: Add New Monitor -> ONVIF -> my cam and login/pass -> stream -> save
Nothing returns to the add monitor window.
Apache log:

> [Sun Aug 12 22:40:44.483447 2018] [php7:notice] [pid 7726] [client 93.80.246.221:57464] PHP Notice:  unserialize(): Error at offset 0 of 460 bytes in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 176, referer: http://site/zm/index.php
> [Sun Aug 12 22:40:44.483513 2018] [php7:warn] [pid 7726] [client 93.80.246.221:57464] PHP Warning:  Invalid argument supplied for foreach() in /usr/share/zoneminder/www/skins/classic/views/monitor.php on line 177, referer: http://site/zm/index.php

in this place (monitor.php on line 177), I put print_r($_request);
and this is what he will translate into the main window:
`Array ( [view] => monitor [mid] => 0 [step] => [probe] => YTo4OntzOjg6IkZ1bmN0aW9uIjtzOjc6Ik1vbml0b3IiO3M6NDoiVHlwZSI7czo2OiJGZm1wZWciO3M6NDoiSG9zdCI7czozOToiaHR0cDovLzEwLjEwLjEwLjE4L29udmlmL2RldmljZV9zZXJ2aWNlIjtzOjQ6IlNPQVAiO3M6NDoiIDEuMiI7czo0OiJOYW1lIjtzOjI0OiJISUtWSVNJT04lMjBEUy0yQ0QyMDMyLUkiO3M6NToiV2lkdGgiO3M6NDoiMTkyMCI7czo2OiJIZWlnaHQiO3M6NDoiMTA4MCI7czo0OiJQYXRoIjtzOjk2OiJydHNwOi8vYWRtaW46WW90YUxqY25AMTAuMTAuMTAuMTg6NTU0L1N0cmVhbWluZy9DaGFubmVscy8xP3RyYW5zcG9ydG1vZGU9bWNhc3QmcHJvZmlsZT1Qcm9maWxlXzEiO30= [saveBtn] => Save )`

[zmonvif-probe.txt](https://github.com/ZoneMinder/zoneminder/files/2281281/zmonvif-probe.txt)

NAME="Ubuntu"
VERSION="18.04.1 LTS (Bionic Beaver)"

zmaudit.pl -v
1.31.45

wpte, 2018-08-14:
I have the same issue with the same setup tested with a couple of browsers (even Edge!)

tdragonite, 2018-09-03:
Same issue, same setup :-(

wpte, 2018-09-21:
Seems to be fixed in 1.32.0

Cramac, 2018-09-21:
no, the problem is in 1.32.

wpte, 2018-09-21:
> no, the problem is in 1.32.

Are you sure? I just downloaded the latest update from Isaac Connor's PPA: https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-master
I tried adding a new monitor, used the ONVIF method and the settings were entered automatically.
First time for me in the master ppa I could do this. Perhaps it's partially fixed?

Cramac, 2018-09-21:
The source is the same for ubuntu. maybe the cache should be cleaned like that (which creates zm)?

knight-of-ni, 2018-09-21:
> [Sun Aug 12 22:40:44.483447 2018] [php7:notice] [pid 7726] [client 93.80.246.221:57464] PHP Notice: unserialize(): Error at offset 0 of 460 bytes in [...]

This error has been fixed in commit f53533e61ed8f23258bed1cd430d63f2ea2290bf

closed 2018-09-21 by knight-of-ni