ZoneMinder/zoneminder #2180: Can't filter for some log events in the log viewer

Warudo, 2018-08-14:
**ZoneMinder Version (`zmaudit.pl -v`):** 1.31.45
**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** Yes, 3dc424282b60ae10ea9f0bde1d36579a1e27e977
**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** Arch Linux
**Browser:** Firefox 61.0.2

When using the Log viewer on the web interface, I can't filter for certain things. For example in the attached screenshot, I can't filter for line 50 to see logins. I can't filter for component web_php, PID 4949 or file auth.php file either. This happens to some other components too, for example I can't filter for line 40, or any other properties of zms:mX lines either.
This is a regression, it used to work in 1.30.4.

When setting LOG_LEVEL_FILE to INFO, no log file web_php gets created either, although all the zm*.log files show up.

Warudo, 2018-08-29:
The LOG_LEVEL_FILE problem is solved. I just had to log in again after enabling it, I though that ZM also writes existing log entries from the DB to the files.

The filtering problem ist not solved however is not solved. The browser console shows no errors.

connortechnology, 2018-09-09:
I think I have this issue fixed or at least improved.

Warudo, 2018-09-29:
Fix confirmed for me.

closed 2018-10-03 by connortechnology