ZoneMinder/zoneminder #2181: Filters related to Storage doesnt work in 1.35.45

qpaz, 2018-08-15:
as already mentioned in #2133 some weird stuff going on... so to double check things i closed #2133, build a new version with empty database, all dependencies double checked ..... and configured zm from zero...(single server env)

now i have two storage areas configured: 
| Id | Path                   | Name    | Type  | Url  | DiskSpace   | Scheme | ServerId | DoDelete |
+----+------------------------+---------+-------+------+-------------+--------+----------+----------+
|  1 | /data/camsurv/events   | Default | local | NULL | 63324019652 | Medium |        1 |        1 |
|  2 | /storage/camsurvarchiv | Archive | local | NULL |           0 | Medium |        1 |        1 |

1st:
When a filter configured as follows:
.... Storage is Default .... it doesnt work in web with an exception and in log with

zm_debug.log.39715 <==
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND ( E.StorageId IS '1' ) order by E.Id asc limit 0,100', ignoring: You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near ''1' ) order by E.Id asc limit 0,100' at line 9]
08/15/2018 10:37:00.735708 zmfilter_23[39715].INF [main:285] [Checking filter TestStorage move returned 0 events ]

2nd:
When a filter configured as follows:
.... Disk Percent greater then or equal to 10 
and StorageArea equal to Default .... 
move all matches to archive, run filter in background, archive all matches
it shows the correct events in web with "List Matches", but on cmd line the filter says 0 events 
==> /var/log/zm/zm_debug.log.8456 <==
       unix_timestamp(E.StartTime) as Time,
       M.Name as MonitorName,
       M.DefaultRate,
       M.DefaultScale
         FROM Events as E
         INNER JOIN Monitors as M on M.Id = E.MonitorId
         LEFT JOIN Storage as S on S.Id = E.StorageId
          WHERE (E.EndTime IS NOT NULL) AND ( 7 >= 10 and E.StorageId = '1' ) and ( E.Archived = 0 ) order by E.Id asc limit 0,100)]
08/15/2018 16:55:24.511880 zmfilter_31[8456].INF [main:285] [Checking filter Test_2 archive, move returned 0 events ]

i have filter reload delay to 120, run audit checked, but for the tests here i restarted zoneminder too, checked if no zm processes are hanging and very often i have to delete zmaudit.pl in /var/run/zm/ manually ...

./zmaudit.pl -v
1.31.45

git rev-parse HEAD
15a6eb7e784ed126256044f125cb6e6ca4962d2e

NAME=Gentoo
ID=gentoo
PRETTY_NAME="Gentoo/Linux"
ANSI_COLOR="1;32"
HOME_URL="https://www.gentoo.org/"
SUPPORT_URL="https://www.gentoo.org/support/"
BUG_REPORT_URL="https://bugs.gentoo.org/"

Maybe some guys out there can reproduce this?

connortechnology, 2018-08-22:
Unfortunately the "IS" comparison is for things like comparing to NULL, not for equality to a numeric value.  Change your filter to use equals instead of IS.

closed 2018-08-22 by connortechnology

qpaz, 2018-08-31:
There was a 2nd issue and i think there is some difference between zmfilter and web storage calculation. look at the screenshot
[image]
 @connortechnology