ZoneMinder/zoneminder #2182: Unable to edit existing Filters or create new Filters

ljr55555, 2018-08-17:
**ZoneMinder Version (`zmaudit.pl -v`):**  1.31.45

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** Originally seen with zoneminder-1.31.45-1.13.20180809gitg037abd491.fc28.x86_64.rpm, also reproduced with zoneminder-1.31.45-1.21.20180814gitg15a6eb7e7.fc28.x86_64.rpm from zmrepo.zoneminder.com. Each installed via RPM

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** Fedora release 28
Linux server.domain.ccTLD 4.17.12-200.fc28.x86_64 SMP Fri Aug 3 15:01:13 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:** n/a

**Relevant log lines:** 
web_js.log:08/16/18 21:23:24.017462 web_js[19516].ERR [ReferenceError: saveFilter is not defined] at http://server.domain.ccTLD/zm/index.php?view=filter line 1
web_js.log:08/16/18 21:23:37.235763 web_js[5761].ERR [ReferenceError: saveFilter is not defined] at http://server.domain.ccTLD/zm/index.php?view=filter line 1

**Steps:** From ZoneMinder Console (classic skin), click "Filters". Select filter from drop-down, change filter settings (modified command to run on "Execute command on all matches"), click "Save As" or "Save"

**Expected Behavior:** Reopening filter tab reflects changes; changes are updated into Filters table within MySQL database

**Actual Behavior:** Filter settings do not reflect changes. MySQL Filters table is unchanged.

connortechnology, 2018-08-17:
I am tempted to think that the issue is caused by an earlier javascript error.  Is there anything listed in the browser's console?  Right click -> Inspect Element -> click on console tab

ljr55555, 2018-08-17:
Thank you, that's exactly the problem. I was getting an access denied error on the JS files served from /zm/cache ... we don't use SSL on our server, and I'd missed commenting out an SSLRequireSSL directive in the new apache config file.

closed 2018-08-17 by ljr55555