ZoneMinder/zoneminder #2184: zmwatch.pl - Shared data size conflict in shared_data

intc, 2018-08-21:
**ZoneMinder Version (`zmaudit.pl -v`):** v1.31.45

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** 24f8d63b6992f114c21697eeebaf615656a6db5a

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):**
Gentoo Linux
hardened/linux/amd64/no-multilib (stable)
gcc version 7.3.0 (Gentoo Hardened 7.3.0-r3 p1.4)
sys-libs/glibc-2.27-r6:2.2
Perl: This is perl 5, version 24, subversion 3 (v5.24.3) built for x86_64-linux

**Relevant log lines:**
```
Aug 21 15:05:42 rec01 zmwatch[6229]: ERR [Shared data size conflict in shared_data for monitor cam01, expected 608, got 875967096]
Aug 21 15:05:42 rec01 zmwatch[6229]: INF [Restarting capture daemon for cam01, shared data not valid]
Aug 21 15:05:42 rec01 zmdc[6152]: INF ['zma -m 1' sending stop to pid 20674 at 18/08/21 15:05:42]
Aug 21 15:05:43 rec01 zma_m1[20674]: INF [zma_m1] [Got signal 15 (Terminated), exiting]
Aug 21 15:05:43 rec01 zmdc[6152]: INF ['zma -m 1' exited normally]
Aug 21 15:05:43 rec01 zmdc[6152]: INF ['zmc -m 1' sending stop to pid 20670 at 18/08/21 15:05:43]
Aug 21 15:05:43 rec01 zmc_m1[20670]: INF [zmc_m1] [Got signal 15 (Terminated), exiting]
Aug 21 15:05:43 rec01 zmc_m1[20670]: INF [zmc_m1] [Unable to read packet from stream 0: error -1414092869 "Immediate exit requested".]
Aug 21 15:05:43 rec01 zmc_m1[20670]: WAR [zmc_m1] [Return from Capture (-1), signal loss]
Aug 21 15:05:43 rec01 zmdc[6152]: INF ['zma -m 1' starting at 18/08/21 15:05:43, pid = 21311]
Aug 21 15:05:43 rec01 zmdc[21311]: INF ['zma -m 1' started at 18/08/21 15:05:43]
Aug 21 15:05:43 rec01 zmc_m1[20670]: ERR [zmc_m1] [Failed to capture image from monitor 1 cam01 (1/1)]
Aug 21 15:05:43 rec01 zmdc[6152]: INF ['zmc -m 1' exited normally]
Aug 21 15:05:43 rec01 zma_m1[21311]: ERR [zma_m1] [Got empty memory map file size 0, is the zmc process for this monitor running?]
Aug 21 15:05:43 rec01 zmdc[6152]: INF [Starting pending process, zmc -m 1]
Aug 21 15:05:43 rec01 zmdc[6152]: INF ['zmc -m 1' starting at 18/08/21 15:05:43, pid = 21315]
Aug 21 15:05:43 rec01 zmdc[21315]: INF ['zmc -m 1' started at 18/08/21 15:05:43]
Aug 21 15:05:43 rec01 zmdc[6152]: ERR ['zma -m 1' exited abnormally, exit status 255]
Aug 21 15:05:43 rec01 zmc_m1[21315]: INF [zmc_m1] [Starting Capture version 1.31.45]
Aug 21 15:05:43 rec01 zmc_m1[21315]: INF [zmc_m1] [Priming capture from http://admin:YWRtaW4=@10.102.5.10:8080/stream/video/mjpeg]
Aug 21 15:05:43 rec01 zmc_m1[21315]: WAR [zmc_m1] [Option -loglevel not recognized by ffmpeg]
Aug 21 15:05:43 rec01 zmc_m1[21315]: WAR [zmc_m1] [Option rtsp_transport not recognized by ffmpeg]
Aug 21 15:05:43 rec01 zmc_m1[21315]: INF [zmc_m1] [Stream open http://admin:YWRtaW4=@10.102.5.10:8080/stream/video/mjpeg, parsing streams...]
Aug 21 15:05:44 rec01 zmc_m1[21315]: INF [zmc_m1] [cam01: images:4 - Capturing at 100.00 fps, capturing bandwidth 267509bytes/sec]
```
This only happens when an recording event occurs. 

Continuous recording seems to fail after a couple of seconds. Monitor mode seems to operate as expected.

connortechnology, 2018-08-22:
I have seen this several times.  As near as I can tell ffmpeg corrupts memory. I havn't yet been able to track down a fix.

intc, 2018-08-22:
Thank you for your comment.

The initial version of ffmpeg was: **media-video/ffmpeg-3.3.6**.
I upgraded it to **media-video/ffmpeg-3.4.4** which presented similar behaviour to 3.3.6.
Just for the fun of it I also tested with **media-video/ffmpeg-4.0.2**. With this version zma returned always with exit status 127.

Small test via cli, both 3.3.6 & 4.0.2 produced a playable video file with command:
`$ ffmpeg -i "http://admin:YWRtaW4=@10.102.5.10:8080/stream/video/mjpeg" -vframes 1000 out.avi`

So no "simple" fix for this one I guess.

intc, 2018-08-24:
My previous comment on ffmpeg-4.0.2 is garbage. After rebuilding ZoneMinder against the 4.0.2 version the system presented similar behaviour to 3.4.4 & 3.3.6.

ASoTNetworks, 2018-09-22:
I have compiled v1.32.0 on Debian and looks like I am seeing the same issue. I found that using H264 camera passthrough will not cause this. 

The event video is missing the timestamp as it bypasses ffmpeg it seems?

ASoTNetworks, 2018-09-27:
I have made the changes to /usr/bin/zmdc.pl and so far recordings haven't stop.

intc, 2018-10-01:
Built today v.1.32.0 (613d88593058055c946359b76cf9e362dfde11cc) / ffmpg-4.0.2. Works fine now! Thank you!

dac2020, 2019-03-15:
I just found this so figured I should give some input. 
Just built a new system on Ubuntu 18.04 server, running 1.32.3 and also noticed this error in the logs.

It says the resource has become `unavailable` yet I am monitoring the same camera from another server and it is never missing a beat.here are two cameras, The 'garage' one and a 'back yard' one. 

The back yard one is the one acting up. It is a cheap Chinese cam.
INS-HD360A-5MP-4X (bought on ebay).

Inesun IP PTZ Bullet Camera, 5 Megapixel Super HD 2592x1944p Security Surveillance Camera Outdoor / Indoor, 2.8-12mm Varifocal Lens, Autofocus, 4X Zoom, 120ft IR Night Vision, IP66, H.265, Onvif 2.4, Motion Detection.

*PTZ IP Bullet Camera Outdoor / Indoor
*ONVIF2.4;Support P2P function
*Support H.264 /H.265 compression
*Pan/Tilt/Zoom, A-B scan, tour available
* Compatible with HIKVISION ,DAHUA, XM NVR
*Compatible with Milestone, NUUO and AXXON platform

Specification
Camera
Sensor: 1/3" CMOS 5MP AR0521
Output pixel: 2592x1944@15fps / 2592x1520@22fps
Lens: 4X f=2.8~12MM
BLC: Support
AGC: Auto/manual
WB: Auto/manual/indoor/outdoor
S/N Ratio>50dB

Is is some of the logging.
`
Mar 15 21:22:35 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:39700 - Capturing at 25.00 fps, capturing bandwidth 7842487bytes/sec]
Mar 15 21:22:38 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:39800 - Capturing at 33.33 fps, capturing bandwidth 10424337bytes/sec]
Mar 15 21:22:42 zoneminder zmc_m7[8180]: ERR [zmc_m7] [Unable to send packet at frame 212: Resource temporarily unavailable, continuing]
Mar 15 21:22:42 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:39900 - Capturing at 25.00 fps, capturing bandwidth 7816403bytes/sec]
Mar 15 21:22:42 zoneminder zmc_m7[8180]: ERR [zmc_m7] [Unable to send packet at frame 212: Resource temporarily unavailable, continuing]
Mar 15 21:22:42 zoneminder zmc_m7[8180]: message repeated 4 times: [ ERR [zmc_m7] [Unable to send packet at frame 212: Resource temporarily unavailable, continuing]]
Mar 15 21:22:45 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40000 - Capturing at 33.33 fps, capturing bandwidth 10432579bytes/sec]
Mar 15 21:22:47 zoneminder zmwatch[1448]: INF [Restarting capture daemon for ptz back-yrd, time since last capture 6 seconds (1552684967-1552684961)]
Mar 15 21:22:48 zoneminder zmdc[1373]: INF ['zmc -m 7' sending stop to pid 8180 at 19/03/15 21:22:48]
Mar 15 21:22:48 zoneminder zmc_m7[8180]: INF [zmc_m7] [Got signal 15 (Terminated), exiting]
Mar 15 21:22:49 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40100 - Capturing at 25.00 fps, capturing bandwidth 7823360bytes/sec]
Mar 15 21:22:49 zoneminder zmdc[1373]: INF ['zmc -m 7' exited normally]
Mar 15 21:22:52 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40200 - Capturing at 33.33 fps, capturing bandwidth 10420224bytes/sec]
Mar 15 21:22:56 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40300 - Capturing at 25.00 fps, capturing bandwidth 7817733bytes/sec]
Mar 15 21:22:57 zoneminder zmfilter_1[1432]: INF [Checking filter PurgeWhenFull delete returned 0 events ]
Mar 15 21:22:57 zoneminder zmfilter_2[1438]: INF [Checking filter Update DiskSpace update disk space returned 0 events ]
Mar 15 21:22:58 zoneminder zmwatch[1448]: ERR [Memory map file '/dev/shm/zm.mmap.7' does not exist.  zmc might not be running.]
Mar 15 21:22:58 zoneminder zmwatch[1448]: INF [Restarting capture daemon for ptz back-yrd, shared data not valid]
Mar 15 21:22:59 zoneminder zmdc[1373]: INF [Command 'zmc -m 7' removed from pending list at 19/03/15 21:22:59]
Mar 15 21:22:59 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40400 - Capturing at 33.33 fps, capturing bandwidth 10418786bytes/sec]
Mar 15 21:23:03 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40500 - Capturing at 25.00 fps, capturing bandwidth 7814898bytes/sec]
Mar 15 21:23:06 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40600 - Capturing at 33.33 fps, capturing bandwidth 10442316bytes/sec]
Mar 15 21:23:09 zoneminder zmwatch[1448]: ERR [Memory map file '/dev/shm/zm.mmap.7' does not exist.  zmc might not be running.]
Mar 15 21:23:09 zoneminder zmwatch[1448]: INF [Restarting capture daemon for ptz back-yrd, shared data not valid]
Mar 15 21:23:10 zoneminder zmdc[1373]: WAR [Can't find process with command of 'zmc -m 7']
Mar 15 21:23:10 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40700 - Capturing at 25.00 fps, capturing bandwidth 7817322bytes/sec]
Mar 15 21:23:10 zoneminder zmdc[1373]: INF ['zmc -m 7' starting at 19/03/15 21:23:10, pid = 8192]
Mar 15 21:23:10 zoneminder zmdc[8192]: INF ['zmc -m 7' started at 19/03/15 21:23:10]
Mar 15 21:23:11 zoneminder zmc_m7[8192]: INF [zmc_m7] [Starting Capture version 1.32.3]
Mar 15 21:23:11 zoneminder zmc_m7[8192]: INF [zmc_m7] [Priming capture from rtsp://"admin":"11112222"@192.168.1.101:554/1/h264major]
Mar 15 21:23:12 zoneminder zmc_m7[8192]: INF [zmc_m7] [Stream open rtsp://"admin":"11112222"@192.168.1.101:554/1/h264major, parsing streams...]
Mar 15 21:23:13 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40800 - Capturing at 33.33 fps, capturing bandwidth 10426136bytes/sec]
Mar 15 21:23:13 zoneminder zmc_m7[8192]: INF [zmc_m7] [ptz back-yrd: images:1 - Capturing at 50.00 fps, capturing bandwidth 109792bytes/sec]
Mar 15 21:23:17 zoneminder zmc_m6[7504]: INF [zmc_m6] [Garage: images:40900 - Capturing at 25.00 fps, capturing bandwidth 7818840bytes/sec]
Mar 15 21:23:19 zoneminder zmc_m7[8192]: ERR [zmc_m7] [Unable to send packet at frame 71: Resource temporarily unavailable, continuing]
`