ZoneMinder/zoneminder #2187: zmdc.pcl always pegged at 100% when camera(s) are not available

vmix44, 2018-08-22:
Running ZM version 1.31.45

I have 3 cameras. If any one of them goes offline, after some minutes, zmdc.pl maxes out at 100% CPU and stays that way indefinitely.

This is also the case when disabling the cameras that are temporarily unavailable. Even when disabling them, the logs indicate the ZM is still trying to access them (why???). Eventually, ZM stops logging errors and warning for those cameras and that is when zmdc.pl maxes out CPU usage.

These cameras are not always available on the network. ZM used to work fine on about a year-old version (don't recall which version, unfortunately). Now, zmdc.pl maxes out CPU usage consistently once a camera goes down. Disabling the camera and restarting ZM has no affect - as described.

Any suggestions would be greatly appreciated.

Thanks

connortechnology, 2018-08-22:
Duplicate of #2121 

We are still waiting for anyone to provide logs.

closed 2018-08-22 by connortechnology

vajonam, 2018-08-23:
thought I could repro this .. but now am struggling! not sure yet, will leave debug logs on for a day and see it comes back..

vajonam, 2018-08-23:
Here are some logs. 

https://www.dropbox.com/s/nwlakjdtzpkfuaw/zm_debug.tar.gz?dl=0. Happened over night, should metion that I have a cron job that looks for zmdc.pl status to see if its running and if not restarts it, I used this with 1.30.4 but not sure if that is adding to the problem. I have turned this off now and will try it again.

vajonam, 2018-08-23:
few more logs

https://www.dropbox.com/s/sjys81l8ah0jb48/zm_newlogs.tar.gz?dl=0

vmix44, 2018-08-24:
One thing I observed that for me, after it stayed pegged for several hours, going to the zm login page yielded a SQL error that there were too many connections.

connortechnology, 2018-08-24:
I have failed at recreating this.  I set my max connections below what is needed to create a reconnection problem. 

However, I can imagine that since we set the mysql_auto_reconnect flag, it might be looping forever in the perl DBI code and thus not logging anything.  Maybe we shouldn't be setting that flag...

Maybe try commenting out lines 117-119 of /usr/share/perl5/ZonemInder/Database.pm
$dbh->{mysql_auto_reconnect} = 1;
      Fatal('Can\'t set mysql_auto_reconnect on in database connection')
        unless $dbh->{mysql_auto_reconnect};

vajonam, 2018-08-26:
Had another issue where some of my cameras stopped recording or zmc stopped and zdmc does not restart it. at the moment I have to a restart script everyday to restart the zm docker container

https://www.dropbox.com/s/n416vmjufpp0xjd/debug_zmc_stopped_logs.tar.gz?dl=0

vmix44, 2018-09-06:
It has gotten worse for me now. Same cameras since the issue began (not added anything new) - Load is pegged between 1.2 and 1.4 - server is hardly unusable. Same process: zmdc.pl. All cameras are running though - no disconnects. Restarting ZM several times and by about the time it finally warms up the third (and final camera), load exceeds 1.0. Then within a minute it is maxed out and stays maxed out.

I would happily share logs, and anything that could help. But there isn't anything logging that is a warning or and error. All cameras are successfully capturing (modect). If there is anything I can do to help, please let me know.

It *seems* like only when all of the cameras come online immediately when ZM starts, does it stay operating in load range of .2 - .5. Once something goes out or is slow to respond, it spins up and stays pegged until restarting. My case right now may be that I am having some latency issues on the network and when ZM starts and it causes the same problem.

vmix44, 2018-09-06:
I don't know if this is useful or not, but after completely being unable to bring up ZM without zmdc.pl maxing out, I disabled each monitor, then restarted ZM. It came up without any problems and the monitors all became Green and with low load. Then, one-by-one, I enabled them (they are all modect), and everything came up okay and the load is within a good range..... (until a monitor becomes unreachable due to a network issue).