ZoneMinder/zoneminder #2188: Existing events lost upon upgrade from stable to master

mnoorenberghe, 2018-08-25:
**ZoneMinder Version (`zmaudit.pl -v`):** 1.31.46

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu bionic main

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** PRETTY_NAME="Ubuntu 18.04.1 LTS"

I had hundreds of archived videos but they were all lost (not displayed in the UI) upon upgrading to master. Perhaps some filter ran that deleted the videos? This happened some months ago but I thought it was just a UI bug on master not showing the archived events but I looked in the DB and they aren't there either.

connortechnology, 2018-08-27:
I can't help but wonder if zmaudit is doing the deleting.  I can't think of any situation that would cause your filters to match archived events.  Can you post what your filters look like?

derSchreiber, 2018-09-15:
I had the same problem in june with the StorageAreas PPA and have restored the deleted events from a backup.
Linux Distribution: "Ubuntu 16.04.5 LTS"

Here is a example log output:
>2018-07-07 12:58:32.886370	zmaudit		10537	ERR	Unable to unlink '/var/cache/zoneminder/events/1/15/08/11/.2652': No such file or directory	zmaudit.pl	
>2018-07-07 12:58:32.759900	zmaudit		10537	INF	Deleting event 2652 from Monitor 1 StartTime:2015-08-11 20:37:19	zmaudit.pl	
>2018-07-07 12:58:32.717520	zmaudit		10537	INF	deleting	zmaudit.pl	
>2018-07-07 12:58:32.683270	zmaudit		10537	INF	Database event '1/2652' does not exist at /var/cache/zoneminder/events/1/15/08/11/20/37/19 in filesystem, deleting	zmaudit.pl	
>2018-07-07 12:58:32.650230	zmaudit		10537	WAR	Path not found for Event 2652 at /var/cache/zoneminder/events/1/15/08/11/20/37/19	zmaudit.pl

/var/cache/zoneminder/events/1/15/08/11/.2652 is linked to 20/37/19
/var/cache/zoneminder/events/1/15/08/11/20/37/19/ is filled with the captured jpgs

StorageScheme in the ZM UI is Medium, in the "Events" database table it is Deep:
> 2652 | 1 | 0 | Alarm-2652 | Motion | 2015-08-11 20:37:19 | 2015-08-11 20:37:59 | 1280 | 720 | 40.51 | 37 | 7 |   | NULL | 7 | 1 | 1 | 1 | 0 | 0 | 0 | 0 | 0 | Motion: xx | 0 | 5823968 | Deep | 0 | NULL

In early 2018 I bought a new camera, so I have changed the monitor from "SaveJPEGs" to "H264 Camera Passthrough"
The only filter running in background is "Update DiskSpace"

As a workaround I have disabled "OPT_FAST_DELETE" and "RUN_AUDIT"

mnoorenberghe, 2018-09-17:
> I can't help but wonder if zmaudit is doing the deleting. I can't think of any situation that would cause your filters to match archived events. Can you post what your filters look like?

Sorry, somehow this email got filtered out of my inbox… 

I see my storage is at 93% now so maybe the PurgeWhenFull filter automatically ran but that is supposed to limit to 5 IIUC so I don't see why all would have been deleted.

[image]

connortechnology, 2018-09-17:
Not saying this is happening to you, but if OPT_FAST_DELETE is on, then disk space is NOT freed by deleting the event (happens in background zmaudit).  But if zmaudit isn't running frequently enough or is just too busy doing other things, then the delete filter can run again before any space is freed, which deletes another 5 events.  This can just keep happening until all events are deleted (although they will still be on disk).  Then zmaudit runs and deletes the files.

This is why I recommend NOT using OPT_FAST_DELETE

connortechnology, 2018-09-17:
I've been reading the code and trying wrap my head around @derSchreiber 's log entries above.  I'm a little stumped.  There are in essence two checks for the files existing and this is saying that both did not find the files.  Yet he says they exist.  

@derSchreiber can you give us some 
ls -l /var/cache/zoneminder
ls -l /var/cache/zoneminder/events/1/15/08/11/

derSchreiber, 2018-09-17:
@connortechnology Here is the output:

`ls -l /var/cache/zoneminder`
`total 52`
`drwxr-xr-x 2 www-data www-data 36864 Sep 15 13:43 cache`
`drwxr-xr-x 3 www-data www-data  4096 Apr 16  2017 events`
`drwxr-xr-x 2 www-data www-data  4096 Jul  9 16:22 images`
`drwxr-xr-x 2 www-data www-data  4096 Dez 10  2015 temp`

`ls -l /var/cache/zoneminder/events/1/15/08/11/`
`total 12`
`drwxr-xr-x 6 www-data www-data 4096 Aug 11  2015 00`
`drwxr-xr-x 4 www-data www-data 4096 Aug 11  2015 01`
`drwxr-xr-x 3 www-data www-data 4096 Aug 13  2015 20`

`ls -l /var/cache/zoneminder/events/1/15/08/11/.2652`
`lrwxrwxrwx 1 www-data www-data 8 Jul  8 00:15 /var/cache/zoneminder/events/1/15/08/11/.2652 -> 20/37/19`

connortechnology, 2018-09-19:
@derSchreiber Would you be willing to let me take a look at your system, using teamviewer or something?  You can contact me privately either at isaac@zoneminder.com or through the chat built into the zoneminder.com website.

derSchreiber, 2018-09-20:
@connortechnology I had a deeper look at my backup data and apt.history and the behavior seems to change from version 1.31.44~20180621110157-xenial to 1.31.44~20180625161604-xenial.
So I will give the old zmaudit.pl a try and if it doesn't work, I will contact you.

connortechnology, 2018-09-20:
I believe I have found the problem.  I have pushed a potential fix to master.  Packages are building. Please test and confirm.

mnoorenberghe, 2018-09-20:
Ah, ok, I did have OPT_FAST_DELETE on.

derSchreiber, 2018-09-21:
Now it deletes the data and leaves the "Events" table untouched.
So events show up in ZM event list, but they are no longer viewable.

Log output:
`2018-09-21 11:55:27	zmaudit		4023	INF	Deleting event directories with no event id information at 1/15/08/11/20/37/19	zmaudit.pl`
`2018-09-21 11:55:27	zmaudit		4023	INF	deleting	zmaudit.pl`

Version: 1.32.0~20180920105855-xenial

connortechnology, 2018-09-21:
Here's an ide.a Turn on debugging so that we can actually see what it is doing.

derSchreiber, 2018-09-21:
Sorry, I only set the Log_Level to debug and forgot the debug switch.

Next try with debugging turned on [zm_debug.log](https://github.com/ZoneMinder/zoneminder/files/2405687/zm_debug.log)

connortechnology, 2018-09-21:
I have pushed some further checks and fixes, please try again.

derSchreiber, 2018-09-22:
It's working now.
Thanks!

closed 2018-10-03 by connortechnology