ZoneMinder/zoneminder #2189: Events are always played in Replay mode regardless of the dropdown setting

mnoorenberghe, 2018-08-25:
**ZoneMinder Version (`zmaudit.pl -v`):** 1.31.46

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** master snapshot from: http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu bionic main

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** Ubuntu 18.04.1 LTS

In the UI I have replay set to None on an event but the single event still replays. Below the player controls it shows "Mode: Replay"

npm-zms is getting the query parameter `replay=none` but the video still loops. This also affect zm-ninja because I think the bug is with the zms side, not the web front-end.

connortechnology, 2018-09-28:
I believe I have fixed this in latest master.  The trick was that it didn't work for jpeg storage, only .mp4 storage.  Should work for all now.

mnoorenberghe, 2018-09-29:
Awesome, thanks! I'll give it a try now

mnoorenberghe, 2018-09-29:
It works now! Thanks.

Fixed by fe444218cc687cbfaa9f2eb49dce0b815476e526 I assume.

closed 2018-09-29 by mnoorenberghe