ZoneMinder/zoneminder #2193: Crud Submodule has Broken RefID

N3X15, 2018-08-28:
## Summary
The `web/api/app/Plugin/Crud` submodule has a broken reference to a commit that doesn't exist.  This was probably caused when Crud's developers force-pushed something and messed up the tree.

## Commit
```
root@zoneminder:~/zoneminder# git rev-parse HEAD
718bb6bba30928ba6275fcfb66a74b0fb2a69e4f
```

## Environment
```
root@zoneminder:~/zoneminder# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 17.10
Release:        17.10
Codename:       artful
```

## Log
```
root@zoneminder:~/zoneminder# git submodule update --init --recursive
error: Server does not allow request for unadvertised object 0bd63fb464957080ead342db58ca9e01532cf1ef
Fetched in submodule path 'web/api/app/Plugin/Crud', but it did not contain 0bd63fb464957080ead342db58ca9e01532cf1ef. Direct fetching of that commit failed.
```

## Fixes
* `git submodule update --init --recursive --remote -- web/api/app/Plugin/Crud` will pull down the latest commit on the branch, but may break things if the upstream version has significant API changes.
* `cd web/api/app/Plugin/Crud; git fetch --all; git checkout <commit>` may work, as well.

knight-of-ni, 2018-08-29:
Try:
```
git submodule sync
git submodule update
```
The problem is actually with the local repo on your disk.

Git can be troublesome when we modify an existing submodule to point somewhere else, which is what we did recently. We see the same issue when we are pushing out own commits. Git basically does not re-sync the submodule on your local repo for you. You were lucky. Usually it just silently fails and continues to use the old submodule without telling you.

The commit in the error message, `0bd63fb464957080ead342db58ca9e01532cf1ef`, is our commit, which lives in our repo: https://github.com/ZoneMinder/crud/commits/v3.1.0-zm

This contains a php-7 compatibility patch. The patch has since been merged into the Crud project but the last time I checked they did not bump the version.

closed 2018-09-21 by knight-of-ni

N3X15, 2018-09-21:
Yep, my bad.