ZoneMinder/zoneminder #2195: Viewing events across monitors stops when 2 events start at the same time.

poldon, 2018-08-30:
When I click an event number (i.e. the total number of events for the past hour on all monitors/cameras) and then click the first event to go to the player page, the Next and/or Previous button will be disabled when I hit an event that started at the same time as another event on a different camera.

Here's an example set of events (Note the event numbers are out of order when they start at the same time):
[image]

If I try to play the first one, the Next button is disabled:
[image]

This makes reviewing a long sequence of events very time consuming.  It took me a while to track down the issue.  Hopefully, it's an easy fix.

**ZoneMinder Version (`zmaudit.pl -v`):** 1.31.47

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** package from zoneminder-master PPA

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** Ubuntu 18.04.1 LTS

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:** N/A

**Relevant log lines:**
```
None
```

connortechnology, 2018-09-04:
Can you confirm that this is still an issue? I have been unable to make it happen and I have recently made a change to how events jump from one to another.

poldon, 2018-09-12:
Yes, it's still an issue.  I just updated to 1.31.47~20180909174905-bionic.

poldon, 2018-09-26:
Have there been any updates on this?  It might be a while before I can try and troubleshoot it myself.

connortechnology, 2018-09-26:
Please note that the latest version is 1.32.0.  

My system handles this case perfectly.  I have push a small fix to sort by Id as well as time, so now the Id's should be in the correct order.

connortechnology, 2018-09-30:
Apparently I am not as smart as I think I am.  Building new packages now which should fix it.