ZoneMinder/zoneminder #2196: Download MP4 button only downloads first event in sequence.

poldon, 2018-08-30:
When I play through a series of events by hitting Next, the Download MP4 button still only downloads the video from the first event.  I am using MP4 (H.264 passthrough) as the storage, not JPEGs.  I would expect the button to send me the event I'm currently viewing, but it always just downloads the first one I clicked on from the event list.

**ZoneMinder Version (`zmaudit.pl -v`):** 1.31.47

**Are you using a development snapshot / git checkout?  If so, what is the latest commit? (`git rev-parse HEAD`):** package from zoneminder-master PPA

**Linux Distribution and Version (`cat /etc/os-release` or `cat /etc/redhat-release`):** Ubuntu 18.04.1 LTS

**If the issue concerns a camera, provide the make, model, frame rate, resolution and ZoneMinder Source Type:** N/A

**Relevant log lines:**
```
None
```

closed 2018-08-31 by connortechnology

connortechnology, 2018-08-31:
I have just pushed a change that reloads the entire page instead of all the trickery we used to do.  It updates the MP4 download link.

poldon, 2018-09-12:
Awesome!  I verified that it works now.  Thank you so much!

commit 9ec415b3 (2018-08-31): Just reload the event view instead of dynamically updating the stream. Fixes #2196