ZoneMinder/zoneminder #2212: h264 camera passthrough causes events to be unviewable

howlett, 2018-09-18:
When selecting storage->h264 camera passthrough, alarms occur but the events are not viewable.

version 1.32.0

No error is logged, the video screen shows the alarm bar (red/not) but cannot load the video.

connortechnology, 2018-09-19:
Are you sure?  With .mp4 the current event will not be viewable (because it is still writing the file) but older events will be.  

What happens if you click Download MP4? Is that file viewable in vlc?

howlett, 2018-09-19:
No, I'm wrong.  For some reason none of the recordings worked for when I had set this up but when I just tested with setting off the alarm, it worked fine.

howlett, 2018-09-25:
I've posted a request for support that may be related to this issue.  No events over 2 days old have any video files yet the events remain.

howlett, 2018-09-28:
This looks to be a configuration issue causing zma to exit with status 255

closed 2018-09-28 by howlett