ZoneMinder/zoneminder #2213: Event is not stored in /var/lib/zoneminder/events/ if directory is empty

spyjo, 2018-09-19:
Hello,
Before upgrading from 1.30.4 to 1.32.0 (from sources), I have cleaned all my events, and my event folder was empty (/var/lib/zoneminder/events/).
After upgrading, when a new event alarm is detected on 1.32.0, the files are not created in /var/lib/zoneminder/events/ 

Checking the logs in zma_m4.log (the test was on my camera number 4): 

09/19/18 19:02:56.945716 zma_m4[25599].ERR-zm_event.cpp/166 [**Can't mkdir /var/lib/zoneminder/events/4/2018-09-19: No such file or directory**]
09/19/18 19:02:56.945956 zma_m4[25599].ERR-zm_event.cpp/172 [Can't mkdir /var/lib/zoneminder/events/4/2018-09-19/13466: No such file or directory]
09/19/18 19:02:56.946125 zma_m4[25599].ERR-zm_image.cpp/963 [Can't open /var/lib/zoneminder/events/4/2018-09-19/13466/00001-capture.jpg: No such file or directory]

The first "mkdir" is assuming that the directory "4" exists and directly try to mkdir "2018-09-19" in it, so it fails.
I suppose it should first try to create the "4" directory before trying to create the folders in it.

Thanks

knight-of-ni, 2018-09-19:
Please show the output of `ls -lhd /var/lib/zoneminder/events`

It would be helpful to know a little more about how you installed ZoneMinder. You said from sources, but how exactly and on what distro? A non-package based installation isn't going to set folder permissions so this is something you would have to know to do manually.

spyjo, 2018-09-19:
ls -lhd /var/lib/zoneminder/events : 
drwxrwxr-x 3 www-data www-data 4,0K sept. 19 19:33 /var/lib/zoneminder/events

When I try to create the directories (1,2,3,4...7) in advance to avoid this problem, with the right chown www-data, they are somehow deleted after few minutes, I suppose because they are empty.
My "4" directory has now 1 event in it (I made an alarm event just after creating manually the directory), so it stays and events in it are ok.

I have installed Zoneminder on Debian (with nginx) from sources like 2 years ago, I think at version 1.29.0. Since then I upgrade manually at every stable release without any problem ;)

knight-of-ni, 2018-09-19:
This is the kind of problem we would have seen immediately if it were happening to a large segment of users. We need to start from the very beginning to figure out what is different with your system. Only then can we determine how best to resolve it. That means we need to see the exact commands you used to build ZoneMinder, especially the cmake parameters specified on the command line. 

Did you install over the top of the previous version of ZoneMinder or did you manually remove the previous version by hand before issuing make install? Manually deleting the previous version is required as `make uninstall` is no longer supported by Perl's make maker.

spyjo, 2018-09-19:
I have just built and installed over the top of the previous version, like I always do.

Here is my procedure for migrating from 1.30.4 to 1.32.0 : 
I have stopped zoneminder
Downloaded https://github.com/ZoneMinder/zoneminder/archive/1.32.0.tar.gz
Installed some new dependancies : libmp4v2-2 libmp4v2-dev x264 libx264-dev libx264-148
Added https://github.com/ZoneMinder/crud/archive/3.1.0-zm.zip in web/api/app/Plugin/Crud/
Added https://github.com/ZoneMinder/CakePHP-Enum-Behavior/archive/1.0-zm.zip in web/api/app/Plugin/CakePHP-Enum-Behavior
Then for compiling:
./cmakecacheimport.sh /etc/zm.conf
cmake -C zm_conf.cmake .
make && make install
./zmlinkcontent.sh
zmupdate.pl (which was successfull)
Then I updated my nginx configuration based on the new apache configuration file (specially for cache and api). And then created the new directory /var/cache/zoneminder/ with chown www-data
and started zoneminder service again.

I think that's all.

connortechnology, 2018-09-19:
As a work around turn off zm audit. It isn't needed.

spyjo, 2018-09-19:
During the upgrade process, the local storage scheme was set to "medium".
If I look at https://github.com/ZoneMinder/zoneminder/blob/master/src/zm_event.cpp : 
starting at line 157 for the "Storage::MEDIUM" part, we can see that the first mkdir is directly trying to create the date "xxxx-xx-xx" directory inside the monitor id directory, without creating or checking that the monitor id directory really exists.

spyjo, 2018-09-19:
@connortechnology thanks I have disabled it for now, I suppose it it this process which is deleting empty monitor_id folders ?
After disabling zmaudit and then restarted zoneminder, all monitor_id folders are created correctly at the start.

knight-of-ni, 2018-09-19:
Glad you discovered what the problem was. I still don't know why this is only happening on your system at the moment, but that is for @connortechnology and I to figure out.

What I was getting at previously was a problem unique to building from source. Installing over the top of a previous version of ZoneMinder can cause stale files to be left around. That can create a situation where ZoneMinder finds the old subroutines in the stale file before it it finds the new. This does not happen often, but when it does it creates some really hard to troubleshoot problems. The Perl scripts are particularly susceptible to this. In any case, it no longer looks like we are dealing with this kind of problem. In the future, the only way to completely avoid this problem is to completely delete ZoneMidner's files (expect for dB and events) before installing the new version.

You may want to consider setting the cmake variable ZM_CONF_SUBDIR. Since zm.conf lives under /etc on your system, the default location of conf.d is also under /etc, which could potentially conflict with another application. Most distros place zm.conf and the conf.d subfolder into /etc/zm.

spyjo, 2018-09-19:
Yes you're right I will change ZM_CONFIG_DIR and ZM_CONFIG_SUBDIR

knight-of-ni, 2018-09-20:
@spyjo On an unrelated note, I have yet to see a working nginx config with the required changes for 1.32.0. If you are willing, can you put your nginx config somewhere publicly so we can get it added into the project?

spyjo, 2018-09-20:
@knight-of-ni Here is my current nginx config file working with 1.32.0. It's not 100% perfect and could be a little cleaned, but it does the job. Api is working great with zmNinja.

```
server {
        listen m.y.i.p:80;
        server_name myserver.mydomain.com;

        root /var/www/myserver.mydomain.com/web/;
        index index.html index.htm index.php;
        autoindex off;
        client_max_body_size 100M;
        server_tokens off;
        access_log  /var/log/nginx/myserver.mydomain.com-access.log;
        error_log   /var/log/nginx/myserver.mydomain.com-error.log error;

        location /zm/cgi-bin {
                gzip off;
                alias /usr/local/libexec/zoneminder/cgi-bin;

                include /etc/nginx/fastcgi_params;
                fastcgi_param SCRIPT_FILENAME $request_filename;
                fastcgi_pass  unix:/var/run/fcgiwrap.socket;
        }

        location /zm/cache {
                alias /var/cache/zoneminder/;
                try_files $uri $uri/;
        }

        location /zm {
                alias /usr/local/share/zoneminder/www;
                try_files $uri $uri/ /index.html /index.php;
                index index.php;

                location ~ \.php$ {
                        gzip off;
                        if (!-f $request_filename) { return 404; }
                        expires             epoch;
                        include             /etc/nginx/fastcgi_params;
                        fastcgi_param       SCRIPT_FILENAME         $request_filename;
                        fastcgi_index       index.php;
                        fastcgi_pass        unix:/var/run/php/php7.0-fpm.sock;
                }

                location /zm/api/ {
                        gzip off;
                        alias /usr/local/share/zoneminder/www/api;
                        rewrite ^/zm/api(.+)$ /zm/api/app/webroot/index.php?p=$1 last;
                }
        }

        location / {
                try_files $uri $uri/ =404;
		
	       	location ~ \.php$ {
	                fastcgi_split_path_info ^(.+\.php)(/.+)$;
	                fastcgi_pass unix:/var/run/php/php7.0-fpm.sock;
	                fastcgi_index index.php;
	                include fastcgi_params;
	        }
        }
}

```

ecostrip, 2018-10-04:
Hello,

For your information, I have deployed a fresh Zoneminder installation on a fresh Centos 7. I confirm that I have the same issue; zmaudit is removing empty monitor folders and zm is not able to save any new modect event after the folder is removed. 

I disabled the zmaudit but this is very inconvenient...

THanks !

knight-of-ni, 2018-10-04:
@ecostrip As you can see from the status of this issue, the fix for this has already been merged into the master branch, and it is part of the 1.32.1 release. New rpm packages were submitted to RPMFusion on Tuesday, and it is now out of my hands. You can either a) wait for the new package to be placed into RPMFusion testing repo or b) manually download the submitted package and install it yourself. See: http://koji.rpmfusion.org/koji/buildinfo?buildID=8940

ecostrip, 2018-10-05:
Hello Knight of ni. Ok Thanks. By the way there is another but and I don't know if you are aware of but symbolic link with Monitor name are not created anymore; is that a normal behaviour ?