ZoneMinder/zoneminder #2214: Export zip or tar file is invalid after download

spyjo, 2018-09-19:
On latest 1.32.0, I have a problem with the event export.
If I try to generate either a tar.gz or zip file, each time I have an invalid file downloaded.
I have verified the files generated in /var/tmp/zm and they are valid, so the problem appears during the download.
I found out that there is a space added on the top of the downloaded file which make it invalid.
I modified the file views/archive.php which is managing the download, with only adding 1 new line:
ob_clean();flush();
just before the readfile() call.
It permits to flush the output to the browser before sending the file in any case of header or space appearing before the datas, so it solves the problem.

knight-of-ni, 2018-09-20:
Just using ob_clean seems to work for me and has been merged to master. See commit ac9129b  
Tested with both single and multi-file exports.

This is intended as a short term fix. @connortechnology has a development branch that improves the overall experience for exports, particularly when video is being exported.

spyjo, 2018-09-20:
Ok thanks

closed 2018-09-20 by spyjo