ZoneMinder/zoneminder #2218: Near Future: Apache/Nginx will need specific access control headers or zmNinja won't work at all for iOS users

pliablepixels, 2018-09-23:
This is really a zmNinja issue, but given it needs server side permissions, I thought I'd open up an issue here to understand what is the best way to get it into the setup of ZM users who use zmNinja without a lot of headaches.

**Android Users**: Doesn't apply to you. Just laugh it off and shake your head at iOS users.

**Summary:** 
Starting iOS 12 (now released), Apple has [officially deprecated UIWebView](https://developer.apple.com/documentation/uikit/uiwebview?changes=_6), which is an old web view in favor of a much faster WKWebView component. zmNinja so far uses UIWebView, but not that its deprecated, Apple may pull support anytime. I've started a branch to move zmNinja to WKWebView. However, this requires that ZM users add CORS headers or zmNinja won't work at all. (UIWebView did not enforce CORS, while WkWebView does)

**These changes need to go into Apache**

```apache
#zmNinja header permissions. Tweak to your needs
Header set Access-Control-Allow-Credentials true
Header always set Access-Control-Allow-Origin "http://localhost:8080"
Header always set Access-Control-Request-Methods "Authorization"
Header always set Access-Control-Methods "OPTIONS,GET,POST,DELETE,PUT"
Header always set Access-Control-Allow-Headers "X-Requested-With, Content-Type, Authorization, Origin, Accept, client-security-token"
Header always set Access-Control-Expose-Headers "Content-Security-Policy, Location"
Header always set Access-Control-Max-Age "1000"
# The next two are apparently needed or pre-flight will fail, but it seems to work fine without it
# if you see pre-flight conditions failing, enable these
#RewriteCond %{REQUEST_METHOD} OPTIONS
#RewriteRule ^(.*)$ $1 [R=200,L]
```

Question for ZM team:

Since not enabling this will essentially completely break zmNinja, are you folks ok if we add these to your apache config files that you bundle along? Is that the best way?

(Someone will have to craft nginx equivalents)

knight-of-ni, 2018-09-23:
Is there a reason why you want to set the header in the web server config file and not in ZoneMinder's php? Is port 8080 actually going to be used on the zoneminder server or does this value simply need to match what wkwebview expects to see?

pliablepixels, 2018-09-23:
1. Yes, this necessarily needs to be in a web server configuration as CORS is a security feature between a web server in one domain and another domain. If CORS requirements are not met, the web server kicks the request out without any app behind the web server getting access to it.

2. `localhost:8080` is what zmNinja will show as its `Origin` so the web server needs to allow it.

knight-of-ni, 2018-09-23:
> 1. Yes, this necessarily needs to be in a web server configuration as CORS is a security feature between a web server in one domain and another domain. If CORS requirements are not met, the web server kicks the request out without any app behind the web server getting access to it. It can be part of a specific virtual host config that handles ZM, for example.

Are you sure? The following seems to indicate you can:
https://www.kerstner.at/2015/02/enabling-cross-origin-resource-sharing-cors-for-php/

Doing in php would allow us to manage this in one place, and would be transparent to the end user. 

> 2\. `localhost:8080` is what zmNinja will show as its `Origin` so the web server needs to allow it.
Right but does that mean the firewall between the client and server needs to have port 8080 open and/or web server needs to listen on that additional port?

pliablepixels, 2018-09-23:
> Are you sure? The following seems to indicate you can:
> https://www.kerstner.at/2015/02/enabling-cross-origin-resource-sharing-cors-for-php/

Oh. I learn something new everyday.  I'd love to test this. Where in php do we add it?

>
> Right but does that mean the firewall between the client and server needs to have port 8080 open and/or web server needs to listen on that additional port?

No

knight-of-ni, 2018-09-24:
I am not familiar with CORS (I guess I will be), but I knew php could rewrite headers.

I'm not sure exactly where to put it yet, but I'd start looking in the cakephp stuff since this is all through the API.

In the meantime, I can temporarily update the demo site with the Apache config if you need something live to test against.

pliablepixels, 2018-09-24:
Ok, so CORS will affect the following:

1. API layer (which you have pointed out above)
2. streaming & control/ptz commands `nph-zms/` URLS (not part of API)
3. Basic Auth (not part of ZoneMinder)

knight-of-ni, 2018-09-24:
Just so we have something, I have placed the apache config mentioned above into this file with a minimal set of instructions:
https://github.com/ZoneMinder/zoneminder/blob/master/misc/apache-cors.conf

pliablepixels, 2018-10-02:
closing - this is no longer a requirement. CORS will be needed for multi-port however (and doesn't just apply to zmNinja - also applies to browser too)

closed 2018-10-02 by pliablepixels