ZoneMinder/zoneminder #2219: None of tha"add" buttons work.

efhalo, 2018-09-25:
When i try to add camera, user or storage nothing happened. Either with chrome or ms edge.
 
**ZoneMinder Version (`zmaudit.pl -v`): 1.32.0
**Linux Distribution and Version: Debian GNU/Linux buster/sid

> Tue Sep 25 11:22:25.165892 2018] [php7:notice] [pid 29219] [client 44.128.0.3:2280] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 11:22:25.571055 2018] [php7:notice] [pid 29219] [client 44.128.0.3:2280] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 11:22:26.082587 2018] [php7:notice] [pid 29219] [client 44.128.0.3:2280] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 11:22:26.490766 2018] [php7:notice] [pid 29219] [client 44.128.0.3:2280] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 11:22:27.010201 2018] [php7:notice] [pid 29219] [client 44.128.0.3:2280] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 11:22:27.414314 2018] [php7:notice] [pid 29219] [client 44.128.0.3:2280] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 11:22:27.913256 2018] [php7:notice] [pid 29219] [client 44.128.0.3:2280] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> WARNING: ZoneMinder configuration file found but is not readable. Check file permissions on /etc/zm/zm.conf_DIR/zm.conf
> [Tue Sep 25 14:50:19.317801 2018] [php7:notice] [pid 32683] [client 44.128.0.3:6875] INF [Login successful for user "admin"], referer: http://44.128.0.2/zm/index.php?view=console
> WARNING: ZoneMinder configuration file found but is not readable. Check file permissions on /etc/zm/zm.conf_DIR/zm.conf
> [Tue Sep 25 14:50:29.304959 2018] [php7:notice] [pid 29137] [client 44.128.0.3:6884] ERR ['addMonitor' is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 14:50:37.654121 2018] [php7:notice] [pid 29241] [client 44.128.0.3:6885] ERR ['addMonitor' is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 14:50:38.142615 2018] [php7:notice] [pid 29241] [client 44.128.0.3:6885] ERR ['addMonitor' is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> [Tue Sep 25 14:50:38.540438 2018] [php7:notice] [pid 29241] [client 44.128.0.3:6885] ERR ['addMonitor' is not defined], referer: http://44.128.0.2/zm/index.php?view=console
> WARNING: ZoneMinder configuration file found but is not readable. Check file permissions on /etc/zm/zm.conf_DIR/zm.conf
> WARNING: ZoneMinder configuration file found but is not readable. Check file permissions on /etc/zm/zm.conf_DIR/zm.conf
> [Tue Sep 25 14:50:50.022338 2018] [php7:notice] [pid 32685] [client 44.128.0.3:6899] ERR ['createPopup' is not defined], referer: http://44.128.0.2/zm/index.php?view=options&tab=users
> [Tue Sep 25 14:50:51.322265 2018] [php7:notice] [pid 32685] [client 44.128.0.3:6899] ERR ['createPopup' is not defined], referer: http://44.128.0.2/zm/index.php?view=options&tab=users
> [Tue Sep 25 14:50:51.817857 2018] [php7:notice] [pid 32685] [client 44.128.0.3:6899] ERR ['createPopup' is not defined], referer: http://44.128.0.2/zm/index.php?view=options&tab=users
> [Tue Sep 25 14:50:52.309026 2018] [php7:notice] [pid 32685] [client 44.128.0.3:6899] ERR ['createPopup' is not defined], referer: http://44.128.0.2/zm/index.php?view=options&tab=users
> [Tue Sep 25 14:50:52.698312 2018] [php7:notice] [pid 32685] [client 44.128.0.3:6899] ERR ['createPopup' is not defined], referer: http://44.128.0.2/zm/index.php?view=options&tab=users
> [Tue Sep 25 14:50:53.109373 2018] [php7:notice] [pid 32685] [client 44.128.0.3:6899] ERR ['createPopup' is not defined], referer: http://44.128.0.2/zm/index.php?view=options&tab=users

knight-of-ni, 2018-09-25:
> WARNING: ZoneMinder configuration file found but is not readable. Check file permissions on /etc/zm/zm.conf_DIR/zm.conf

Something looks wrong with your build. How did  you build and install ZoneMinder?

efhalo, 2018-09-25:
I installed from  debian repository.

efhalo, 2018-09-25:
>  dpkg -l zoneminder
> 
> ii  zoneminder               2:1.32.0-dmo2     i386              Linux video camera security and surveillance solution

knight-of-ni, 2018-09-25:
In any case, the following path is not right:
/etc/zm/zm.conf_DIR/zm.conf

That gets set during the build process.
ZoneMinder can't find its config file so lots will go wrong. You may want to try to contact the package maintainer.

You could also just build your own package. We have instructions in our readthedocs site.

efhalo, 2018-09-25:
This is it: https://www.deb-multimedia.org/pool/main/z/zoneminder-dmo/zoneminder-dmo

knight-of-ni, 2018-09-25:
Yeah, that's a third party site. It's not a official debian repo.  

Looking at the packaging scripts, there is a custom patch being applied called `01_hardcoded-etc.diff` that is mangling where zoneminder looks for its config file. This isn't something we can fix on our end. Only the package manager can.

knight-of-ni, 2018-09-25:
wrong button

efhalo, 2018-09-25:
i try with changed config path - that error is gone, but everything is the same.

efhalo, 2018-09-25:
> [Tue Sep 25 17:21:36.495915 2018] [php7:notice] [pid 1427] [client 192.168.1.154:8980] INF [Login successful for user "admin"], referer: http://192.168.1.13/zm/index.php?view=console
> [Tue Sep 25 17:21:40.252238 2018] [php7:notice] [pid 1411] [client 192.168.1.154:8983] ERR [Uncaught ReferenceError: addMonitor is not defined], referer: http://192.168.1.13/zm/index.php?view=console
> [Tue Sep 25 17:23:11.864798 2018] [php7:notice] [pid 32686] [client 192.168.1.154:8991] ERR [Uncaught ReferenceError: createPopup is not defined], referer: http://192.168.1.13/zm/index.php?view=options&tab=users

onlyjob, 2018-09-26:
@efhalo, deb-multimedia.org is not an official Debian repository. It's best to avoid it. It is even mentioned in the FAQ [here](https://wiki.debian.org/DebianMultimedia/FAQ#There_is_.27Debian_Multimedia_Maintainers.27_and_.27deb-multimedia.org.27._So_what.27s_the_difference.3F) and [here](https://wiki.debian.org/DebianMultimedia/FAQ#Well.2C_isn.27t_deb-multimedia.org_an_official_part_of_Debian.3F).

knight-of-ni, 2018-09-26:
@efhalo - @onlyjob is the package maintainer for the zoneminder package that lives in the official debian repos. He has been coordinating with us to get the latest ZoneMinder release into a working Debian package.

Note that we plan to release 1.32.1 soon. This will  be a bug fix release, and I'd recommend waiting for it.

While you wait for an official package to show up, you can always build your own ZoneMinder package using the instructions in our readthedocs pages.

efhalo, 2018-09-26:
@knight-of-ni Thank you! I drop the dmo repo, and reinstall previous version, it is works fine.