ZoneMinder/zoneminder #2220: 1.32.0: Camera stopped working (Image size is not multiples of 12 and 64)

onlyjob, 2018-09-26:
Some cameras stopped working with error `Image size is not multiples of 12 and 64` after upgrade to 1.32.0.

Here is one example: http://www.weathercamnetwork.com.au/new_farm/982x737.jpg - easy to configure as HTTP/simple.

That was working fine with 1.30.4 and earlier versions.

onlyjob, 2018-10-09:
Any progress here? 1.32.1 didn't fix it and it would be really nice to restore feed from not perfectly sized cameras... Thanks.

onlyjob, 2018-10-09:
There was a similar issue here before: #979

connortechnology, 2018-12-08:
I did some work on this... just got side tracked..

onlyjob, 2018-12-13:
Thanks, @connortechnology. I've just upgraded to 1.32.3 but this problem is still present...

closed 2019-02-22 by connortechnology

strusty, 2019-03-23:
You may want to look at the video itself in a browser, copy the picture, paste it into gimp, and see the size that gimp reports.  It may be different than the 982x737 or other odd size that is reported at that link.  This is the case when using Android application IP Camera.   Inspecting the size at browser will give a wrong size.  If you use the size gimp reports, it will work and be a multiple of 64.

PR #2404 (2019-02-22): Fix imagesize requirements
Fixes #2220 

When constructing an Image object, test width and height to see if we can use the fast delta and blend functions.  Fall back to slow functions for odd images.