ZoneMinder/zoneminder #2228: 1.30.4 on gentoo:getting SIGSEGV in zmc

igelskalar, 2018-10-01:
segmentation fault occurs in
Monitor *Monitor::Load( unsigned int p_id, bool load_zones, Purpose purpose ) @zm_monitor.cpp
in line 2707:unsigned int id = atoi(dbrow[col]); col++;
dbrow[col] is undefined in this context

I was able to trace it down to
MYSQL_ROW zmDbFetchOne( const char *query ) @zm_db.cpp
111:MYSQL_ROW dbrow = mysql_fetch_row( result );
112:mysql_free_result( result );
Obviously mysql_free_result now frees up all memory connected to result including all dbrow content. I think this is the right behaviour for the Mysql-API and mysql_fetch_row only returns pointers to the resultset memory.
A gdb trace reveals that after reaching line 113, the result and dbrow variables are undefined as expected.
But I am a little bit astonished about the behaviour of the mysql-API now freeing all connected memory to the results despite the old versions are not cleaning up a derived MYSQL_ROW from a result set.
My installed gentoo mysql packages are:
dev-db/mysql-5.6.40-r2
dev-db/mysql-connector-c-6.1.11-r1

connortechnology, 2018-10-03:
I'm afraid you are going to have to update to 1.32.1 and let us know if the problem persists.  I bet it won't as I rewrote most of that code.

igelskalar, 2018-10-08:
I saw your rewritten code already bevor opening the issue, but I though there would be the possibility to repair the stable tree to make my distro working again. Now I switched to the Master branch, compiled successfully from code and brought this version up and running. Thank you so far!

closed 2018-10-08 by igelskalar