ZoneMinder/zoneminder #2229: Api method getDiskPercent throw corrupted json

zhuykovkb, 2018-10-02:
I use the latest build:
```
root@3175d0de62c1:/ZoneMinder# zmaudit.pl -v
1.32.1
root@3175d0de62c1:/ZoneMinder# git rev-parse HEAD
d2c2fbbfb979a567d348e24c7bd43a6002d1f745
root@3175d0de62c1:/ZoneMinder# 
```

Runnig in Docker, but it's no matter.
```
root@3175d0de62c1:/ZoneMinder# lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.5 LTS
Release:        16.04
Codename:       xenial
root@3175d0de62c1:/ZoneMinder# 
```

I try to use Api (https://zoneminder.readthedocs.io/en/stable/api.html#host-apis). That methods works properly:
```
zhuykovkb@home:~$ curl -XGET  https://zm.beget.ru/api/host/daemonCheck.json
{
    "result": 1
}zhuykovkb@home:~$ 
zhuykovkb@home:~$ curl -XGET  https://zm.beget.ru/api/host/getLoad.json
{
    "load": [
        5.03,
        4.83,
        4.72
    ]
}zhuykovkb@home:~$ 
zhuykovkb@home:~$ 
```
But when i try to use this, i have:
```
zhuykovkb@home:~$ curl -XGET  https://zm.beget.ru/api/host/getDiskPercent.json
<pre class="cake-error"><a href="javascript:void(0);" onclick="document.getElementById('cakeErr5bb3db33b33fe-trace').style.display = (document.getElementById('cakeErr5bb3db33b33fe-trace').style.display == 'none' ? '' : 'none');"><b>Notice</b> (8)</a>: Undefined index: ZM_DIR_EVENTS [<b>APP/Controller/HostController.php</b>, line <b>114</b>]<div id="cakeErr5bb3db33b33fe-trace" class="cake-stack-trace" style="display: none;"><a href="javascript:void(0);" onclick="document.getElementById('cakeErr5bb3db33b33fe-code').style.display = (document.getElementById('cakeErr5bb3db33b33fe-code').style.display == 'none' ? '' : 'none')">Code</a> <a href="javascript:void(0);" onclick="document.getElementById('cakeErr5bb3db33b33fe-context').style.display = (document.getElementById('cakeErr5bb3db33b33fe-context').style.display == 'none' ? '' : 'none')">Context</a><pre id="cakeErr5bb3db33b33fe-code" class="cake-code-dump" style="display: none;"><code><span style="color: #000000"><span style="color: #0000BB">&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;</span><span style="color: #DD0000">'fields'&nbsp;</span><span style="color: #007700">=&gt;&nbsp;array(</span><span style="color: #DD0000">'Name'</span><span style="color: #007700">,&nbsp;</span><span style="color: #DD0000">'Value'</span><span style="color: #007700">)</span></span></code>
<code><span style="color: #000000"><span style="color: #0000BB">&nbsp;&nbsp;&nbsp;&nbsp;</span><span style="color: #007700">));</span></span></code>
<span class="code-highlight"><code><span style="color: #000000"><span style="color: #0000BB">&nbsp;&nbsp;&nbsp;&nbsp;$zm_dir_events&nbsp;</span><span style="color: #007700">=&nbsp;</span><span style="color: #0000BB">$zm_dir_events</span><span style="color: #007700">[</span><span style="color: #DD0000">'ZM_DIR_EVENTS'&nbsp;</span><span style="color: #007700">];</span></span></code></span></pre><pre id="cakeErr5bb3db33b33fe-context" class="cake-context" style="display: none;">$mid = null
$zm_dir_events = array()</pre><pre class="stack-trace">HostController::getDiskPercent() - APP/Controller/HostController.php, line 114
ReflectionMethod::invokeArgs() - [internal], line ??
Controller::invokeAction() - CORE/Cake/Controller/Controller.php, line 491
AppController::invokeAction() - APP/Plugin/Crud/Lib/CrudControllerTrait.php, line 31
Dispatcher::_invoke() - CORE/Cake/Routing/Dispatcher.php, line 193
Dispatcher::dispatch() - CORE/Cake/Routing/Dispatcher.php, line 167
[main] - APP/webroot/index.php, line 107</pre></div></pre>{
    "usage": {
        "726-1 cam1": {
            "space": "0",
            "color": "red"
        },
        "726-3 door": {
            "space": "0",
            "color": "red"
        },
        "727 table": {
            "space": "0",
            "color": "red"
        },
        "726 storage room": {
            "space": "0",
            "color": "red"
        },
        "727 kicker": {
            "space": "0",
            "color": "red"
        },
        "727 coffeemaker": {
            "space": "0",
            "color": "red"
        },
        "726 trainee1": {
            "space": "0",
            "color": "red"
        },
        "726 trainee2": {
            "space": "0",
            "color": "red"
        },
        "725 cam1": {
            "space": "0",
            "color": "red"
        },
        "725 cam2": {
            "space": "0",
            "color": "red"
        },
        "724 cam1": {
            "space": "0",
            "color": "red"
        },
        "724 cam2": {
            "space": "0",
            "color": "red"
        },
        "717 cam1": {
            "space": "0",
            "color": "red"
        },
        "726-1 cam2 new": {
            "space": "0",
            "color": "red"
        },
        "726-3 nights new": {
            "space": "0",
            "color": "red"
        },
        "727 door new": {
            "space": "0",
            "color": "red"
        },
        "Total": {
            "space": "48.230037689209",
            "color": "#F7464A"
        }
    }
}zhuykovkb@home:~$ 

```

If i should paste another information, tell me, please.

knight-of-ni, 2018-10-03:
Confirmed.
It looks like this function needs to be re-written to use the diskspace values being written to the dB rather than call `du` for each monitor.

knight-of-ni, 2018-10-03:
Note the API call to get totaldiskspace for a single monitor works:
`http://localhost/zm/api/monitors/{monitor-id}.json`

The field in the json data you are looking for is called TotalEventDiskSpace.

connortechnology, 2018-10-03:
Looking into this, the function is really bad.  I can't really fix it without changing it drastically.  Are you prepared for it's output to be vastly different than it was before? I'm kinda hoping that you are the only person in the world using this function.

First off, this function returns disk usage, not percentage.  
Second if you specify a monitor it returns it in human format (112K) instead of in megabytes (.112)
Further, we now support multiple storage areas, so really this function needs to return a complete list of every storage area and how much of each storage area monitors are taking up.

I am wondering if the data you want can be gotten through the other models, and we could maybe delete this function.

zhuykovkb, 2018-10-03:
I'm sorry that still don't describe my position. I really don't need this function. I've tested ZM API over update to the latest release, only for test, and found a bug. If this is really useless function i think it can be removed on the next release.

closed 2018-10-03 by knight-of-ni

juanmoura, 2019-04-15:
When I run getDiskPercent it returns only the default directory value of zoneminder. But it does not return from the other disks.

connortechnology, 2019-04-17:
getDiskPercent is going to be deprecated.  Use the Storage api calls to get info on each storage area, or the monitors object to get usage by monitors.

PR #2232 (2018-10-03): Fix 2229 get disk percent
Fixes #2229 

Please note that this also fixes one inconsistency. When specifying mid, the output was in human format (112K for example).  This now returns it without the K, in Kb.

This does not fix the inconsistency that if mid is not specified, the outputs are in Mb.

Further it does not fix the fact that the data returned is NOT a percentage of anything.  Further, since we can now have multiple storage areas, a percentage would be meaningless unless broken down into per-storage area data.

A better way to get this data is to query the monitor or storage area or ev [... cut]