ZoneMinder/zoneminder #2251: H264 Passthrough, FPS drop when audio enabled

rajil, 2018-10-13:
I have a Dahua HDW5231R-ZE camera which supports audio with codecs (G.711A, G.711Mu, G.726 and AAC). I have using the G.711A option.

With the audio disabled ZM Console shows the expected Capture 10.00/10.00 FPS. However, when the audio is enabled this drops to Capturing 5.00/4.76 fps. 

I have confirmed using VLC that the camera is indeed transmitting video at 10fps even audio is enabled. ZM is somehow reducing the fps.

connortechnology, 2018-10-26:
Encoding the audio to aac and writing it out to disk is more work for your cpu to do.  It may be that your cpu is just not capable of keeping up.

ASerbinski, 2018-10-26:
@connortechnology : Is the audio encoded even in passthrough mode? I would think that it would be best only to alter the audio stream when using X264 encoder, and let the audio pass through when in passthrough mode. Either that, or add a separate tick to control audio compression.

connortechnology, 2018-10-26:
Yes.  A .mp4 can only hold AAC.  If your camera will send AAC then you should configure it to do so.  

In a future version you will have a choice of .mkv which can hold anything.

rajil, 2018-10-26:
> Encoding the audio to aac and writing it out to disk is more work for your cpu to do. It may be that your cpu is just not capable of keeping up.

I dont see the CPU pegged to 100% when audio is enabled. Running mediainfo on the generated mp4 shows that the FPS is indeed 10.

closed 2019-01-02 by stale[bot]

gurka, 2025-09-20:
> In a future version you will have a choice of .mkv which can hold anything.

@connortechnology 

Sorry for bumping this old issue and tagging you, but I'm very interested in being able to switch to mkv to be able to passthrough both video (h264) and audio (pcm_alaw). Is this available yet?

In the documentation I see that there should be a "Recording" tab in the monitor settings page, where you should be able to set output container, but I can't find it in my installation (v1.36.35). Do I need to use the unstable/test version 1.37 or wait for 1.38?

connortechnology, 2025-10-02:
Yes, you will require 1.37+