ZoneMinder/zoneminder #2253: montagereview.js contains hardcoded links to example.com/zm

Nocifer, 2018-10-14:
OS: Arch Linux (not that it matters in this case)
ZoneMinder version: Confirmed as still present in master branch

In its default configuration ZM runs its Web UI from a 'zm' subdirectory, e.g. example.com/zm. That's fine, but I (and I'm sure many other people) prefer to run my servers with a domain- or port-based setup, i.e. zm.example.com or example.com:1234

The problem is that, when ZM is run this way (or even when it's run from a different subdir), Montage Review (and possibly other parts of ZM that I haven't yet stumbled upon) does not work properly because it's currently hardcoded to assume ZM is being run from the subdir 'zm'. The culprits are lines 127 and 135 in _WEBDIR/skins/classic/views/js/montagereview.js_ (I have **bolded** the relevant parts):

<pre><code>
124 return location.protocol + '//' + server.Hostname + 
125   //'/cgi-bin/zms?mode=jpeg&replay=single&event=' + event_id +
126   //'&frame='+Frame.FrameId +
127   '<b>/zm</b>/index.php?view=image&eid=' + Frame.EventId + '&fid='+Frame.FrameId +
...
...
...
135 return '<b>/zm</b>/index.php?view=image&eid=' + Frame.EventId + '&fid='+frame_id + "&width=" + monitorCanvasObj[monId].width + "&height="
</code></pre>

Removing '/zm' from these two lines is an easy enough fix, but unfortunately it results in the same problem when trying to do the reverse: if ZM is run from a 'zm' subdir, as is the default, then the link fails once more because now it _lacks_ that '/zm' part - not to mention, the issue with running from a differently named subdir is not addressed in either case.

One way to fix this would be to change the subdir part of the link from a hardcoded to a dynamic value and have some pre-install script fetch the WEBDIR path, extract the correct subdir (if any) and insert it to the relevant lines. So the above code block would become something like this:

<pre><code>
124 return location.protocol + '//' + server.Hostname + 
125   //'/cgi-bin/zms?mode=jpeg&replay=single&event=' + event_id +
126   //'&frame='+Frame.FrameId +
127   <b>SUBDIR_VALUE</b> + '/index.php?view=image&eid=' + Frame.EventId + '&fid='+Frame.FrameId +
...
...
...
135 return <b>SUBDIR_VALUE</b> + '/index.php?view=image&eid=' + Frame.EventId + '&fid='+frame_id + "&width=" + monitorCanvasObj[monId].width + "&height=" +
</code></pre>

It goes without saying that this is just me taking a quick look and trying to think of a quick fix, without having properly debugged, nor being properly familiar with, the codebase.

connortechnology, 2018-10-15:
We have a PR that addresses this, not sure when it will be ready though.

In the meantime, you can just add an alias in your apache config so that /zm works as well.

The reason is that configuring everything everywhere to use something other than /zm is very confusing for most people, so if we just make the backend urls work with /zm and the front be whatever and have both work, then things are easier for new users to configure.

Nocifer, 2018-10-17:
OK, thanks for the reply!

aaronsta1, 2018-11-11:
im getting this error in 1.32 as i changed my SSL port to 8443 in the default-ssl.conf 
i dont understand the fix.. putting a link in /zm to go to /www/??

can you be more specific?

connortechnology, 2018-12-10:
That PR was merged and released in 1.32.3.

Nocifer, 2018-12-11:
Are you talking about #2335 (I saw it mentioned in #2336)? Yes, the issue has been partly fixed (e.g. no more malformed URL calls in live or montage view) but as you previously said, the backend still retains many instances of hardcoded /zm/whatever calls, mainly (but not solely) API and CSS calls as far as I can see, so I guess the 'zm' -> 'www' link or alias is still needed.

By the way, I noticed this while skimming through _skins/classic/includes/export_functions.php_:

```
628    echo "<div><a href=\"javascript:switchevent('$eids/zm/EventImages.html');\"> $eids </div>"; 
```
but
```
633    <div><a href="javascript:switchevent('<?php echo   $eventPath[$eid]; ?>/zmEventImages.html');"><?php echo$eid?></a></div>
...
645    <div><a href="javascript:switchevent('<?php echo   $eventPath[$eid]; ?>/zmEventImages.html');"><?php echo$eid?></a></div>
```

Shouldn't it be either 'zmEventImages.html' or 'zm/EventImages.html' in all three cases..?

knight-of-ni, 2019-01-02:
So I believe the fix for this is to create a new var in montagereview.js.php:
`var PathToIndex = <?php echo Server->PathToIndex?>;`

Then replace `/zm/index.php` at line 132 in montagereview.js with `PathToIndex+ {the rest of the string}`

knight-of-ni, 2019-01-03:
fixed in #2377

closed 2019-01-03 by knight-of-ni