ZoneMinder/zoneminder #2261: Force Alarm inside monitor page and fps count , 1.32.1 was working, 1.32.2 not working - BUG - when on virualhost

multijohn, 2018-10-18:
Force Alarm inside monitor page and fps count , not working with 1.32.2 !
It was fixed on 1.32.1 !

Ubuntu 16.04

Force Alarm does not do anything, and fps just displaying "- fps"!

In Chrome's developer tools i am seeing the following error !

"Failed to load http://homeserver/zm/index.php/zm/index ... command=99: Response to preflight request doesn't pass access control check: No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://homeserver:8085' is therefore not allowed access."

 I am running ZM on a virtual host (: 8085), and the problem was created by upgrading to 1.32.0 version, disappeared with 1.32.1, and came back with 1.32.2 !

Thank you  ZM team !

connortechnology, 2018-10-18:
I think I have this fixed in master already, will go into 1.32.3.

SmokeyBR, 2018-10-18:
is this a small fix ? if so could you point us to the commits or changes so we can try it ?

connortechnology, 2018-10-19:
Actually now I'm thinking it was still broken in master. Try out the latest commit:
https://github.com/ZoneMinder/zoneminder/commit/e7ec2faccda692735cf2f7739d606eaf41400891

connortechnology, 2018-10-19:
I've applied yet another fix for when running on a non-standard port.  Please try the latest.

SmokeyBR, 2018-10-19:
i can confirm with these changes applied this issue has been resolved
https://github.com/ZoneMinder/zoneminder/commit/e7ec2faccda692735cf2f7739d606eaf41400891
https://github.com/ZoneMinder/zoneminder/commit/f9b1453705fef14aa9c67193c2477f74eb739ed6

Nocifer, 2018-10-20:
Yup, I can confirm the fix resolves the erroneous Ajax calls. Nice job. But this issue also affects the Zone view, namely at lines 474 and 549 in skins/classic/views/js/zone.js, which should also be changed accordingly.

SmokeyBR, 2018-10-20:
http://domain.name.ex:5101/zm/index.php i use "subfolder" "zm"

gkthemac, 2018-10-21:
@Nocifer I too have non-functional playback control buttons below the video, with a 500 error in the apache logs when I click them. I'm sure if this is caused by a "dumb" camera or not, since all my cameras are Local i.e. analogue composite video connected to a video capture card and accessed via /dev/video*.

This issue also affects replay of past events, but only when the camera source is set to write each frame as an individual JPEG, with the Video Writer turned off. If I turn off Save JPEGs and set the Video Writer to x264 Encode then replay of events recorded from that point on do have working playback controls, with the added "benefit" of not being able to replay the old JPEG-saved events.

But even with x264 Encode switched on, the live camera views still have non-functional playback controls which do nothing except log 500 errors.

lbdroid, 2018-10-25:
Those commands have nothing to do with smart vs. dumb cameras.
They're just of the nature "play", "pause", "query", etc. Defined here;
https://github.com/ZoneMinder/zoneminder/blob/master/web/includes/config.php.in#L106

lbdroid, 2018-10-25:
FYI: The problem with command=99 is caused by this commit;
https://github.com/ZoneMinder/zoneminder/commit/bc6d8623a6ef7e41cf5134c07c6cd8f449d977b7

It crashes on sem_get(..), which is *NOT* universally included in php builds.
http://fr2.php.net/manual/en/sem.installation.php

Workaround for it is to add the following near the top of web/ajax/stream.php:
```
if (!function_exists('sem_get')) {
    function sem_get($key) {
        return fopen(__FILE__ . '.sem.' . $key, 'w+');
    }
    function sem_acquire($sem_id) {
        return flock($sem_id, LOCK_EX);
    }
    function sem_release($sem_id) {
        return flock($sem_id, LOCK_UN);
    }
}

if( !function_exists('ftok') )
{
    function ftok($filename = "", $proj = "")
    {
        if( empty($filename) || !file_exists($filename) )
        {
            return -1;
        }
        else
        {
            $filename = $filename . (string) $proj;
            for($key = array(); sizeof($key) < strlen($filename); $key[] = ord(substr($filename, sizeof($key), 1)));
            return dechex(array_sum($key));
        }
    }
}
```

connortechnology, 2018-10-25:
Ok I will add these.  

Which distro doesn't compile in semaphore support?

ASerbinski, 2018-10-25:
Fedora 25 doesn't. I haven't had the opportunity to look at any other RH, like newer releases or EL.

Nocifer, 2018-10-26:
Yup, this was the culprit, everything works correctly now. Thanks @ASerbinski !

Arch also does not enable the extension by default, but it's compiled in so it's just a matter of enabling it. Thanks again.

gkthemac, 2018-11-06:
Gentoo only compiles semaphore support into PHP when the "sysvipc" USE flag is set; just done that and now my playback controls are working again.

bradc87, 2018-12-06:
I am also having this issue on Arch Linux using the AUR package. Is there an eta on 1.32.3?

connortechnology, 2018-12-17:
Can we please get an update and have this issue closed if 1.32.3 has fixed this issue?

Nocifer, 2019-01-07:
The issues outlined in the OP are the result of two different things:

The malformed "zm/index.php/zm/index.php" URL was fixed by the commits mentioned by SmokeyBR in https://github.com/ZoneMinder/zoneminder/issues/2261#issuecomment-431506022.

The 500 error and the problems with the feed info (state, fps, mode, zoom) not being available and force alarm not working were a result of missing PHP semaphore support. @ASerbinski posted a fix in https://github.com/ZoneMinder/zoneminder/issues/2261#issuecomment-433129559. I don't know if @connortechnology has merged it or not (can't see anything relevant in ajax/stream.php) but in my current 1.32.3 setup disabling semaphore support results in no errors, so I assume it has indeed been fixed.

I think this issue can be safely closed.

multijohn, 2021-05-11:
Hello,
Issue is back with 1.34.26 on Ubuntu 18.04 php7.3!
Force alarm and cancel alarms buttons not working!

connortechnology, 2021-05-11:
Fix will eventually be released in 1.34.27

thillRobot, 2021-07-12:
I am having the same problem with a fresh install of ZoneMinder 1.34.26 and Ubuntu Server 18.04. Everything seems to be working except the _Force Alarm_ and _Disable Alarms_ buttons. Can you elaborate on how to apply the fix shown above? I assume it is something simple, but I cannot locate the file `watch.js` that is modified in the fix.

thillRobot, 2021-07-18:
I have re-installed my system for a different purpose. I had the disk logical names backwards from what is typical, and I wanted a new host name for the server. Either way, the *Force Alarm* and *Disable Alarms* buttons are working for me again. I assume this means the issue was related to my configuration (or the issue has been fixed by an update).

commit bb6c045a (2021-05-11): Fix query params when including auth_relay. Fixes #2261
closed 2023-01-04 by connortechnology